bambulab/BambuStudio #1339: [Issue] Parameter Field Focus

BambuLabDiscord, 2023-02-25:
This is a *VERY* annoying bug.  After clicking in a field in Bambu Studio, focus should NOT be taken away from that field without a CLICK.  

Existing behavior:  Moving your mouse cursor away from a text/parameter field causes that field to lose focus.

bambu123, 2023-02-28:
The 3D canvas takes the focus if you move the mouse there.

bambu123, 2023-02-28:
@LiZ-Li-BBL Can the 3D canvas not auto-take focus?

LiZ-Li-BBL, 2023-02-28:
In order to use the toolbar functions properly, the focus need to be set to the canvas. It is not an expected behavior that focus be taken away from a field without a mouse click. We will fix it in the near future.

LiZ-Li-BBL, 2023-04-21:
Fixed it in new release v1.6.0 Public Beta.

closed 2023-04-21 by LiZ-Li-BBL