bambulab/BambuStudio #1340: Implement Hole Horizontal Expansion for small holes

amarand, 2023-02-25:
**Is your feature request related to a problem? Please describe.**
When I print small holes (for example: 6.3mm modeled), they are not dimensionally accurate. In order to get a 6.3mm hole, I need to model it as 6.8mm. I've heard this is a common issue with 3D printers and slicers. Cura has a feature called "Hole Horizontal Expansion" which allows you to dial in the hole shrinkage, to get the perfect size holes. For reference, when printing a short (8mm) cylinder with a 80mm diameter, with 10 6.3mm holes, the cylinder thickness and diameter are spot-on. I believe this to be an issue with small holes.

**Describe the solution you'd like**
Implement the "Hole Horizontal Expansion" feature in Bambu Studio, as implemented in Cura.

**Describe alternatives you've considered**
Modeling the holes bigger, but that seems like a sub-optimal solution. 

**Additional context**
I do not know if PrusaSlicer (which Bambu Studio is based on?) has this option, or if it's in a different menu or under a different name. Please note that I do not want to adjust the entire print's scale. Scaling up my entire print to make the holes bigger messes with the other dimensions in the print. This is not a problem with anything but small modeled holes shrinking.

MadMax389, 2023-02-26:
This already exists. Simply enter the correction in mm and it shrinks or expands the holes accordingly

[image]

amarand, 2023-02-26:
Sorry I missed that. Thanks for directing me to that setting!

closed 2023-02-26 by amarand