bambulab/BambuStudio #1342: BUG: Can't add text to painted surfaces

Photogad, 2023-02-25:
**Describe the bug**
BUG: Can't add colored text to painted surfaces

**3mf File for This Bug**
[HowWhyWhatCone.zip](https://github.com/bambulab/BambuStudio/files/10832467/HowWhyWhatCone.zip)

**To Reproduce**
Paint an object multiple colors, then add colored text to different sections (change the text to "modifier" so it only changes color... NOPE doesn't work :(

**Expected behavior**
add colored text to multiple surfaces

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: windows12
 - Latest version

zhimin-zeng-bambulab, 2023-03-06:
We have improved the text tool in v1.5.0 public beta. 
@Photogad  Can you still reproduce this issue in v1.5.0?

zhimin-zeng-bambulab, 2023-03-16:
@Photogad  Hello, does the v1.5.0 Studio meet yout needs?

zhimin-zeng-bambulab, 2023-03-21:
@Photogad 
In studio, the painting color is higher than the color of modifier, so if both modifier and painting are set, only the color of painting will be displayed.
I'm sorry that this question is limited by the program framework and cannot be modified for the time being.

If you want to display the text modifier color, you could cut the object into parts, then set different parts to different colors, and then add a modifier to each parts.

Photogad, 2023-05-14:
@zhimin-zeng-bambulab 
Thank you for your insight. If this is not possible due to program limitations, we can close this issue?

closed 2023-05-15 by zhimin-zeng-bambulab