bambulab/BambuStudio #1344: FEATURE REQUEST: Allow editing text once placed so you don't have to make a new text and re-enter size and position

Photogad, 2023-02-25:
**Is your feature request related to a problem? Please describe.**
FEATURE REQUEST: Allow editing text once placed so you don't have to make a new text and re-enter size and position

**Describe the solution you'd like**
I'd like to be able to edit text once it is placed so that if I want to add multiple texts, I don't have to create a new one and re-do size, rotation, position, etc

**Describe alternatives you've considered**
There is none!

**Additional context**

[image]

MadMax389, 2023-02-26:
Yes please

icerabbitje, 2023-02-26:
Totally.

This is in my top 5 frustrations with Studio. Cannot edit what you did.

Want to print 6 or 12 items with different colored labels? Don't like how something printed ... 

You're in for some hard labor recreating it over and over.  No text edit for even a typo or change is inconceivable really.

MadMax389, 2023-02-26:
As a workaround, I usually name the text object with the font and size so I know what it was if I come back to that project. e.g. Arial-16-2mm. Implementing the OP's request may be harder than it looks once the object is converted to an stl.

icerabbitje, 2023-02-26:
It may not have an immediately simple solution, but the system could save the data components from before the slicing. 

I have been trying to work with text from CAD instead, but gee B Studio is a giant pain in terms of aligning, with everything jumping to the print bed, no distance indication, no auto-aligning other than bed center, ...

MadMax389, 2023-02-27:
If you r-click and add the stl as a part, it won't stick to the bed. No auto aligning but at least you can move it in the z-direction wherever you want.
[image]

Photogad, 2023-05-14:
@LiZ-Li-BBL  @StoneLiBambulab 
I think this has been added in recent version of Bambu Studio? Can you confirm , if so, close this issue?

LiZ-Li-BBL, 2023-05-19:
It has been added.

closed 2023-05-19 by LiZ-Li-BBL