bambulab/BambuStudio #1345: FEATURE REQUEST: Show BOTH Imperial and Metric measurements inside the scale box

Photogad, 2023-02-25:
**Is your feature request related to a problem? Please describe.**
I would like the software to show BOTH metric AND imperial measurements inside the scale box. This is very important for us USA customers who receive measurements from our clients in imperial (inches) measurement, but as 3D printer professional we work with metric. This will make it easy for us to convert!!! Without having to google the conversion every time we need to convert.

**Describe the solution you'd like**
Add extra boxes in the scale window for metric/imperial so both are visible. OR you could add a little "checkbox" there (LIKE PRUSASLICER HAS!!!!) to switch between imperial/metric without having to go all the way to File -> Preferences -> Measurement Units -> OK   every time!!! Very slow that way! And not convenient!

**Describe alternatives you've considered**
The only other alternative is google to convert imperial to metric every time but this is very slow and annoying.

**Additional context**
[image]

LiZ-Li-BBL, 2023-03-06:
Thank you for your suggestion. We will consider this in the future.

Photogad, 2023-05-14:
@LiZ-Li-BBL 
> Thank you for your suggestion. We will consider this in the future.

Thank you, this would be incredibly useful for your American customers.

EklDuderino, 2023-09-03:
I 2nd this addition. Prusa slicer has a check box that toggles on the fly between metric and imperial measurements. This would be a great improvement. Please add this feature. Thank you!

Ali229, 2023-09-28:
From USA and would love this feature as my customers know inches and I do everything in mm. I suggest a simple setting in preferences "Show both units" that would show the main one selected here on top and then another row of size for the unselected.
[image]

hiddenworldhopper, 2024-10-26:
Having just changed over from Prusa to Bambu Studio - this is a very missed feature. Even if this has been open for over a year

L1S9D7M3T, 2025-08-13:
> Thank you for your suggestion. We will consider this in the future.

PLEASE ADD THIS FEATURE! we have been waiting for over a year.