bambulab/BambuStudio #1346: Surface Mode like Cura

robcoleNS, 2023-02-26:
It would be nice to get surface mode similar to Cura's. Very handy and required when printing RC planes.....

Thoemse, 2023-03-05:
Please add this. It is not only for RC planes but for anything that needs to be printed as light as possible.

SaltWei, 2023-03-06:
Current it's available to print RC plane. I had printed one plane from 3DLabPrint and it can fly.
Settings are shown as following.
[image]
[image]
[image]

SaltWei, 2023-03-06:
By the way, if you want to make the model to be closed top. Just add one modifier with certain thickness and set the top layer thickness to be non-zero.
[image]

robcoleNS, 2023-03-06:
Thank you for your advice SaltWei, but for instance when I slice the attached file, there is a spar on the inside of the rudder that disappears with slicing.....(actually I can't seem to attach the file......)

robcoleNS, 2023-03-06:
Picture 1 shows the rudder with the internal spar (for lack of a better term) before slicing.  Picture 2 shows 2 indentations in the end of the rudder where you insert a hinge also before slicing. Picture 3 shows the rudder after slicing with that internal spar gone and picture 4 shows those indentations for the hinges completely removed from the model......
[image]
[image]
[image]
[image]

gfalgiano, 2023-06-05:
@SaltWei other RC plane creators are using the surface mode feature such as planeprint.com

Currently, it only works with Cura and Ideamaker.

Surface mode would be a great addition to Bambu Studio since the Bambu printers are popular with RC enthusiasts.

I attached a Zip file with the sample part, so you can see how it slices in Bambu Studio. Even/odd feature doesn't help with this file. [WINGTEST_PLA_Planeprint.zip](https://github.com/bambulab/BambuStudio/files/11649278/WINGTEST_PLA_Planeprint.zip)

Here is some more info about why it is needed:

**Describe the solution you'd like**
If possible, it would be great to have support for Surface Mode as described here: https://support.makerbot.com/s/article/1667418049490

**Describe alternatives you've considered**
I see that Bambu Studio has picked up "Even-Odd" mode from Prusa slicer to support 3D printed planes from another designer 3D Lab Print https://3dlabprint.com/faq/prusaslicer/ , but the slicer settings don't work for PlanePrint files.

**Additional context**
Similar request for Prusa slicer: https://github.com/prusa3d/PrusaSlicer/issues/6982
One user suggests this feature can benefit academic research into 3D printing tissue for medical applications.

Here are some images of a sample file comparing the model vs sliced results. You can see that the internal structure is missing.

[image]

[image]

xuhao1, 2023-06-14:
+1 I also want this mode!

xuhao1, 2023-07-23:
@SaltWei Any update to this?

SaltWei, 2023-07-31:
The current multicolor scheme is not compatible with the model aircraft. Very hard to support. Can't support it in the short term

BambooBugFixer, 2023-08-04:
Hi, we are analyzing the reason why some RC plane models can be sliced with Even-odd mode while some other RC plane models can be sliced well with surface mode. Could you please tell me what kind of 3 D software you usually use to design RC plane models? This will help us figure out the reason in detail.

robcoleNS, 2023-08-04:
Thank you for continuing to look into this issue. I don't design any plane files myself, I get mine from Plane Pilots, 3d Lab Print, Cults  etc etc. I am however happy to share any files you might need in order to help you figure out the issue

gfalgiano, 2023-08-06:
> Hi, indeed we've noticed that some RC plane models can only be sliced well with Surface mode. But we've also found that some RC plane models can only be sliced well with Even-odd mode while some it doesn't work well with Surface mode. This made us rethink about whether we should add Surface mode in BambuStudio or not. Could you please tell me what kind of 3 D software you usually use to design RC plane models? This would help a lot.

@BambooBugFixer This page may help. It is from one of the popular RC plane designers that uses a method that requires surface mode. Look at the section towards the top and the Profile 3 (P3) area. He offers free sample files well. The intent is to lower the weight of the model for ones that are intended to be printed out of standard PLA.

He has a large back catalog of planes designed with this technique and they require surface mode. I spoke to him over email before, so he may be willing to discuss this with you or the bambu lab team. I'm sure he would appreciate the best printers out there being capable of reproducing his designs :)

You can see from this post and others, it's desired by the community as well. Please let me know if there is anything else I can do to help.

[https://www.planeprint.com/print](https://www.planeprint.com/print)

gfalgiano, 2023-08-06:
@BambooBugFixer I believe they do something in the design software that creates nonmanifold edges, which the slicer doesn't normally like. I believe the goal is to have these lines in the outer skin of the model so close together than they make the slicer draw a continuous line similar to vase mode in some areas and more traditional layers in other areas where different shapes or structure is needed.

Here is an example of what the model and a single layer look like in cura with PlanePrint recommended settings.

[image]

xuhao1, 2023-08-07:
@BambooBugFixer I am trying to print this: https://www.youtube.com/watch?v=ZBwomKoYSZc&t=271s
I must use CURA for slicing (which is REALLY trouble for bambu). Bambu Lab can't provide solid result.

hwinkel, 2023-08-14:
I'm entering the space with Plane prints as well, would be nice to have a proper setup for the typical models. Especially as Bambulab is starting to enter the business with LW-PLA Aero ...

a1axx, 2023-08-29:
Please try to add surface mode! Especially with the new PLA-Aero, we are limited to the models we can print as most are designed to be printed with surface mode.

MaxLansing, 2023-09-30:
Hello everyone, i would like to second that request! Adding this ability, especially since there are no official profiles for Cura for BambuLab printers, this feels like a must. 
TBH, i knew before spending a lot of money on my X1C that there was no out of the box possibility to get RC models printed with it, and this particular part is just part of one of my hobbies, and the printer does a lot of stuff, which others can't.
It would be f*** awesome to print Planes with BambuStudio and BambuLab printers!

Since i am no programmer, i can't help directly, but i would like to know if there is anything the non-coding folk can do to get this feature implemented?
Anything we can help with?

robcoleNS, 2023-10-11:
This is the one feature that is holding me back from buying ANOTHER Bambu lab printer.....

shmoopie, 2024-03-29:
Bump to see if any work has been done on this?  Surface Mode would really open up the ability for printing RC plane models... PLEASE!

mab500, 2024-05-08:
Yes I'd be very happy if the Surface mode is add to the bambu slicer Planprint has a lot of nice models to print

hwiosec, 2024-10-04:
If Bambu Lab want to add ‘surface mode’, they need to change the slice engine, which is libslic3r, as well as the curaengine in Cura. It’s a bit complex. Neither prusa and BamBu hasn’t throughly iterate the core engine in the recent years.

rebelboy, 2025-01-07:
Please add Surface Mode.
We really need it！