bambulab/BambuStudio #1348: Make Bambu Studio editing function better and visually clear like TinkerCad

icerabbitje, 2023-02-26:
**Is your feature request related to a problem? Please describe.**
We all get this printer to do multiple color prints.
We make separate components in CAD, that we want to print together.
But it is next to impossible to actually align and assemble anything in BS. 
The controls are too rudimentary, there is no distance separation shown.
Everything snaps to the build plate.
As soon as you bring anything in, everything jumps off the build plate.

Try to properly align text and images on the bottom of an object in BS, to understand the major frustration that BS causes at this time.  

**Describe the solution you'd like**
The team should have a close look at TinkerCad, for just a half hour, try to bring a few pieces together, move text, align (L), ... 
Nothing snaps to the plate = BS needs a toggle to Snap on/off
Gap to the build plate is shown. 
One can work in mm or partial mm increments so we know stuff aligns.
There's various align functions (L).

Copy those functions and that visual clarity into BS, so we can bring the CAD parts we separated to be printed in different colors, back together into one print.

MadMax389, 2023-02-26:
Assuming that the separate stl's in the CAD model are properly aligned, simply import all the stl's together and BS will ask you if it is a single model. It will maintain the proper relationship between the models.

[image]

[image]

[image]

icerabbitje, 2023-02-26:
Great recommendation!
I just tried that, importing 4 parts at once, and that will help a great deal.
Hopefully the team will still consider improving the editing functions. 
Thank you so much for replying !!! 

Less reason to pull my hair out in Studio ;)