bambulab/BambuStudio #1350: QR no read to damage plate

Benderrodrigu3z, 2023-02-26:
No qr scan oh wait I forgot to add the plate, places plate which changes the Z offset but the machine doesn’t recheck to see if it changed, if it does change remeasure everything if it’s the same continue

your nozzle grooved a brand new plate and now it’s trash please add this check

[image]
[image]

icerabbitje, 2023-02-26:
I think for some reason it checks the plate late.

I just got a new plate, swapped it in but forgot to re-slice, printer gave an alert ... thought about just doing resume; but I hit stop; because measuring the plates with calipers real quick, they're different thicknesses; so you can't just safely say resume, either it digs in the plate or prints in the air ... if it does not recalibrate.  

I think ideally, the printer should say which plate is expected, and resume makes the printer re-check again, then re-level the print bed ( using an off brand bed) and only then print. 

Or simply advise to re-slice with the appropriate plate in Studio.

Benderrodrigu3z, 2023-02-26:
Check Z offset the look for QR code
If no QR have a restart button or resume button and say looking for High temp plate not found or textured plate not found, Continue, Restart and Check Plate, Cancel

SaltWei, 2023-02-28:
Thank you very much for suggestions!
But this seams to be handled by printer and maybe official support can't be got timely here(For slicer).

I have transfered this ticket to printer team.

For future issue about firmware and printer, please move to https://forum.bambulab.com/ or BanbuHandy APP to report ticket (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log) for better support. Thanks!

sodachen22, 2023-04-27:
The user can just resume the printing job after he puts on a new hotbed. Because he knows the hot bed and the new hot bed either is correct or he does want to print with this type of hotbed.