bambulab/BambuStudio #1351: Print started but just sat all night doing nothing

Benderrodrigu3z, 2023-02-26:
I got no notification to my phone on this
Also how is it at 17% done with this error?

the issue here is the qr has an air bubble under it so it can’t read the bump… yall need to add a check above and below by 2mm for QR Code and left to right

[image]

icerabbitje, 2023-02-26:
Part of the printing process, is the toolhead homing, bed leveling, test lines, ... and with a small file, so that could maybe be 17% of the code and/or time. 

I ran into this when I switched plates ( just got the textured one ) and it required a manual press on resume. I actually stopped, and re-sliced it, as I measured the plates were different thicknesses ... just to be safe. 

I guess one of the things Bambu could maybe tweak, is to ensure that this plate check is done as early as possible.

ChrMaass, 2023-03-04:
I also thing it would be best if the check starts first after homing, I‘m running into same issue starting the job for the night and go away and it stoped after a few minutes because of selecting the wrong buildplate. Hope you can change the order to do it first.

ChrMaass, 2023-03-04:
#1187

tangketan, 2023-03-07:
We have already brought forward QR code scanning before bed leveling, right after homing and nozzle wiping. Further bringing it forward is risky, as the Z coordinate may not be accurate, and moving the toolhead above the QR code may cause the nozzle crash with bed.

MrDB42, 2024-05-14:
I would really like to see this moved even further forward in the process. Actually, I would like it to be the second thing done:
1. Home nozzle (get proper Z height)
2. Check build plate type
3. Check build plate alignment
4. Check lidar, camera, etc.
etc.

Reason: To many times I have started a print with the wrong build plate because the downloaded model is set for some other plate that I will be using (i.e. Model: Textured PEI, I use Smooth Cool PLA plate.). The print bed is heated to a temperature different than what is needed, so when I go back to the model, change the build plate type, re-send to printer, I have to wait for it cool down before it prints.

OR, Just plane forget to put a build plate on the print bed. (It happens).

There should be someway to check type, and if present, of build plate as soon as possible before any heating (bed, nozzle), filament loading, nozzle cleaning, etc.