bambulab/BambuStudio #1352: Supports Interface can't be turned off.

randyre, 2023-02-26:
**Describe the bug**
There is no option to print supports without an interface. With the interface, larger portions that have supports create a large surface that sticks to the part and is almost impossible to remove. 

**Workaround**
Use support material as an interface, but this becomes expensive and makes the print take much longer.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable supports
2. attempt to find a way to remove or minimize the support interface.

**Expected behavior**
Option under interface to be "none"

SaltWei, 2023-02-28:
Thanks for reporting issue!
Do you mean the top interface layer doesn't work when set to be 0 layer?

randyre, 2023-03-11:
Yes, even when set to '0" it was generating a support interface. Doesn't seems to happen on all models. I need to go back and find the one I was having trouble with.

randyre, 2023-03-12:
Also would like to have a "0" for the bottom or base.

MusangKingBambulab, 2023-03-24:
hi @randyre , could you please provide any .3mf for me to diagnose the interface issue? When there are top z and no interface layers, support will be generated without interface layers and it works for me just fine. Looking forward to your reply.