bambulab/BambuStudio #1353: Print finish with one missing layer

b-reich, 2023-02-26:
**Describe the bug**
I successful print finish with 30/31 Layer.
Not sure if the last Layer is missing or it is just a ui error.

**3mf File for This Bug**
[Clips.zip](https://github.com/bambulab/BambuStudio/files/10834697/Clips.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Print Project
2. Device --> Print Progress
3. Wait until Print finish
4. End with 30/31 Layers

**Expected behavior**
Print should finish with 31/31 Layers

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: iOS and MacOS

SaltWei, 2023-02-27:
This is known issue that total layer number is wrong.
Has been fixed in internal version and will be released soon.

SaltWei, 2023-03-02:
Fixed in 1.5.0 Public beta version

closed 2023-03-02 by SaltWei