bambulab/BambuStudio #1354: FEATURE REQUEST: "Select All" for media browser items

Photogad, 2023-02-26:
**Is your feature request related to a problem? Please describe.**
I have like over 9000 time-lapse videos on my printer's SD card that I would like to delete. I cannot believe in the media browser there is no "select all". You can presently multiple select, but you have to select one at a time, but that would take way too long. So the only way I could delete them is formatting the SD card which is not a smart or convenient way of deleting the timelapses. Plus, there are about 10 timelapses I want to keep!

**Describe the solution you'd like**
A button to put checkmarks in all of the media items.

**Describe alternatives you've considered**
Formatting the SD card to delete many timelapses, which is annoying and won't let me keep some of them.

**Additional context**
[image]

bambu123, 2023-03-02:
Perhaps we should provide a selection of year/month groups, selecting all files is dangerous because everything is selected.

Photogad, 2023-03-04:
> Perhaps we should provide a selection of year/month groups, selecting all files is dangerous because everything is selected.

That would be better than nothing. Some way to mass-delete videos

psy0rz, 2023-04-25:
check out https://github.com/bambulab/BambuStudio/releases/tag/v01.06.00.65

can probably be closed now.

closed 2023-04-26 by bambu123