bambulab/BambuStudio #1357: Enable or disable dark mode checkbox on macOS

suchabadperson, 2023-02-27:
This option exists in Bambu Slicer for Windows, but is not present in the macOS version and would be very useful.

walterwongbbl, 2023-03-02:
Hi.
Studio's dark mode will follow the system settings of macos. 
modify appearance setting, and studio will follow the system to switch dark mode.
[image]

suchabadperson, 2023-03-02:
Yes, but I find it very useful to be able to switch to  dark mode only within the application, keeping the light theme on macOS.

keithel, 2023-10-14:
This issue also applies to the Linux release of Bambu Studio (AppImage, tested on Ubuntu 23.04 Lunar Lobster). Dark/Light mode is only configurable in the Gnome system configuration dialog. I'd like the ability to switch from dark mode to light mode just within Bambu Studio.

The reason I need this is to work around another bug, which I came here to report - while searching to see if anyone had reported it yet. That bug is that the text color of properties in the "Object" (per-object) pane is extremely illegibile - the color is a grey color that is only barely different from the dark mode background grey. I'll report this separately.

suchabadperson, 2023-10-14:
It does not seem that they will ever implement this idea as it exists in other applications. Unfortunately there is no other way to enable it in macOS than switching everything to dark mode.

closed 2024-01-18 by walterwongbbl

keithel, 2024-01-26:
@walterwongbbl Did you close this because it's now implemented (enabling/disabling dark mode within app/decoupled from window manager setting)? 
I thought I saw an issue go by that was the same as this marked done...

suchabadperson, 2024-03-27:
Any update on this? @walterwongbbl

packhamc, 2025-07-07:
Any news on dark mode for Linux?