bambulab/BambuStudio #1360: Linear Adv on PEI

cryd-s, 2023-02-27:
**Is your feature request related to a problem? Please describe.**
Feature Request for Linear Adv

**Describe the solution you'd like**
It would be an advantage if you could start a calibration process for all filaments stored in the AMS (selection menu), e.g. on the technical plate. After calibration, the values should be transferred to the slicer and only replaced when the filaments are re-sychronized from the AMS.

**Describe alternatives you've considered**
...

**Additional context**
...

TheyWho, 2023-02-28:
THIS is a very viable solution to the Bambu Lab's mysterious method(s) of storing (or not) the various settings and calibrations for each filament.  Right now people are just guessing which parameters are saved for how long.  For instance:  when starting a print that uses two or more filaments, the FIRST one called has a flow test...  so what about the other filaments used in the same part?   Are you using a previous calibration for filaments 2 and 3, or is the machine just winging it?  We need some documentation to go with your great machine.  Do ALL calibrations evaporate when the power is turned off?    Saving the calibration data to a file that is available until a new filament replaces it OR each filament's calibration data can be stored under the UPC code for each product.  Whenever the user wants to pull that filament off the shelf, he/she just finds that UPC number and we are ready to roll.

If this is a NEW UPC number then the system will prompt the user to do a complete calibration suite on that particular filament and then it will be available until the user feels that a re-calibration is warranted.  This could REALLY be a game changer that elevates your terrific system from "a hobby machine"  to something that can be a production workhorse or a very simple to use "greatest toy on earth".

user897943, 2023-03-01:
I concur.