bambulab/BambuStudio #1361: Bambu Studio hangs when selecting play for  camera

julie777, 2023-02-27:
**Describe the bug**
This was working fine, but yesterday when I selected "play" for the camera on my P1P Bambu Studio hung.
I have repeated it many times.
Today I happened to look in the Windows event viewer and found this entry

```
Faulting application name: bambu-studio.exe, version: 1.0.0.0, time stamp: 0x63c8d22a
Faulting module name: D3D9.DLL, version: 10.0.19041.2311, time stamp: 0x86e4ce0b
Exception code: 0xc0000005
Fault offset: 0x000000000000a0cd
Faulting process id: 0x1304
Faulting application start time: 0x01d94ae2ed685300
Faulting application path: C:\Program Files\Bambu Studio\bambu-studio.exe
Faulting module path: C:\WINDOWS\SYSTEM32\D3D9.DLL
Report Id: 564f49a9-4fc2-4b25-bcbf-6d955456284e
Faulting package full name: 
Faulting package-relative application ID: 
```

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Print plate
2. deselect level
3. select send
4. select "play" right arrow button for camera

**Expected behavior**
I would expect the camera image to be shown in the Bambu Studio window on the Device tab.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: 
 ```Edition	Windows 10 Home
Version	21H2
Installed on	‎2/‎27/‎2023
OS build	19044.2673
Experience	Windows Feature Experience Pack 120.2212.4190.0
```

 - Bambu Studio Version: 1.4.2.13

bambu123, 2023-02-28:
Seam relates to D3D9.dll.
Please refer:
https://github.com/bambulab/BambuStudio/issues/244

bambu123, 2023-03-02:
@julie777 Have you had some progress?

julie777, 2023-03-07:
It appears to have been fixed by the 2023-03-02 firmware release.

Thanks.

julie777, 2023-03-08:
I wrote another long reply that seems to have vanished.

I had more hangs when two instances of Bambu Studio were open at the same time and I selected play on the camera. After kill both instances, starting Studio again and hitting play worked.

Now it is a couple hours later and Bambu Studio kept timing out trying to connect to my printer, then it would just close by itself. I finally went to the task manager and killed all Edge and Edge WebView processes and now Studio can connect to my printer.

julie777, 2023-03-08:
I have now tried many things and I can't print. 
I open bambu studio and restore my unsaved model.
I go to device and select my printer and it connects and updates the printer information on the display
I select play on the camera and it shows the image
I slice and then select print plate
The print dialog opens and it says synchronizing device
then it times out
and bambu slicer closes

I have rebooted Windows. 
I have tried again and again. 
The last time it actually sent to the printer and turned on the camera.
Then bambu studio crashed/closed.
I just restarted studio and connected to the printer and I am viewing the camera.

My thoughts...

I think there is a problem with the networking for Studio and the error handling is not catching errors causing it to crash. I don't know of any problems with my network (using LAN mode). Nothing else on my network is having problems.

The Studio network test has no problems, but it doesn't appear to check direct printer connection.

I just remembered that there are log files. 
I am attaching some of the log files, one for each type of ending:
[log.zip](https://github.com/bambulab/BambuStudio/files/10915280/log.zip)

bambu123, 2023-03-08:
Maybe you should update your printer firmware.
We use a TLS connection now.

julie777, 2023-03-08:
I updated my firmware with the P1P release from 2023-03-02. Is there a newer version?

Regardless of the printer firmware, Studio should not crash ever.

bambu123, 2023-03-08:
Can you help try with the zip packet version which can dump the crash log for debugging?

https://github.com/bambulab/BambuStudio/releases/download/v01.05.00.59/Bambu_Studio_win-v01.05.00.59-20230301140620.zip

bambu123, 2023-03-08:
@julie777 You are using the old version 1.4 of BambuStudio. Maybe you should update to the new version.

julie777, 2023-03-08:
I am running the code that you provided. So far it is working.
I did notice that it downloaded the network plugin. Does the network plugin come from BambuLab instead of being part of the Studio code?
I wonder if I just needed to update the network plugin? I don't see anyway in Studio to check the network plugin version.

According to help/check for updates I am running the newest version of Studio. I normally avoid beta versions, especially when I am having so many problems with the release version. I can try 1.5beta if you want me to.

cmguo, 2023-03-08:
Yes, it is a beta version from here. The formal release is coming soon.
https://github.com/bambulab/BambuStudio/releases/tag/v01.05.00.59

bambu123, 2023-03-08:
Yes, it is a beta version from here. The formal release is coming soon.
https://github.com/bambulab/BambuStudio/releases/tag/v01.05.00.59

julie777, 2023-03-09:
I just had it crash with the zipped version you had me install. Here are the logs.
[log.zip](https://github.com/bambulab/BambuStudio/files/10926507/log.zip)

I don't understand why requests need to be made to bambulab.com in order to run Studio and my printer in LAN only mode.

bambu123, 2023-03-09:
It is strange for the new 1.5 version to crash.

Lan Only mode only blocks printer from www network, not Studio.

julie777, 2023-03-10:
I just downloaded the log.zip that I posted and it has the log files in it.

I am running the version that you posted for me "Internal Version 1.5.0.59".

The crash happens in the device view, when manually selecting connect to printer, when trying to start camera, and when updating printer status during preparing to print.

bambu123, 2023-03-10:
I am expecting the files named crash-xxxx.log.
To ensure the  bambu_networking.dll is up to date, remove the 'plugins' directory and let it fetch a new one.

julie777, 2023-03-11:
There was no crash log after that crash. 
I did get a new plugin.

bambu123, 2023-03-14:
So what's the problem with the live view now?
Is it failed to play?
Is it hanging the program and you can not operate everywhere after starting the live view?

Please do not send the printing to print when testing the live view.

julie777, 2023-03-15:
I don't understand. Why is there a problem with having live view on before starting a print?
BTW, that is not the issue because I very seldom have it on.

Sometimes, when not printing, I enable live view to check my other printer that I can see from the camera on my P1P. This works most of the time. Sometimes it causes a hang.

When Studio hangs, I do not get a crash log, even though I am using your internal version. I believe that this is because the crash is happening in the network dll and can't be caught.

I suggest creating a diagnostic app that loads the network dll and exercises the function in it to be able to see what is happening.

Even better would be to move the network dll into the Studio repo so that I could see the source code and help to find the problems.

Question:
When I go to the devices page and select my printer the network icon has some kind of orange mark on it that goes away after Studio makes the connection. What does the orange mark signify?

bambu123, 2023-03-16:
Thank you for the detailed information.
It seems not to be the D3d9 problem because it is sure to hang the program every time you play live view.
Then it hangs sometimes, not crash, so no crash log is generated. 

To analyze the problem, we need you to create the mini-dump when the program is hung.
The mini-dump is created from the Windows Task Manager, by selecting the hung progress and clicking the 'Create Dump file" context menu.
Please make sure the problem is really hung and not responding to mouse move and click, like this one:
[image]

julie777, 2023-03-17:
I will try to capture the mini dump.
First I want to share more information I have collected.
I have tried to make sure that the wifi connection is as good as possible. I turned off 5Ghz on my router and moved my router to sit right next to my P1P. My computer is wired, so the only 2Ghz device on the network is the printer.

Here is the result of pinging the printer.

[image]

I used "Advanced IP Scanner" and I can see the P1P on the network. 

[image]

I tried using Acrylic Wifi Analyzer network traffic and channel use.
The pink is my wifi router and it isn't sharing a channel with nearby routers. 
Note: I am running on my computer which is in a different room from the printer, so the signal strength from the router should be even higher at the printer.
The pink is my wifi, and frequently it is the only signal showing.

[image]

When I view packets on the wifi network all I see is beacon packets from my router and from the neighbor's routers. I don't see any traffic from the printer. This is my first time trying to analyze network traffic so I don't know if I am doing something wrong.

Periodically the printer will disconnect and Studio will stop showing the camera feed. I have to reconnect the printer. I think this indicates a problem with the printer hardware or firmware, unless bambu studio is causing the disconnect. However based on the ping response, and the fact that when I start to print it often times out getting printer details and then works when I refresh it seems more likely to be in how the printer is "connecting" to the network.

Do you have access to the source code for bambu_networking.dll?
What is the bambusource.dll for?

julie777, 2023-03-17:
I upgraded from the internal build you gave me to the latest build when it was released. Do I need to go back to your build to get a usable crash dump?

julie777, 2023-03-17:
While waiting for a hang, I installed wireshark and ran a packet capture test to see what the communication looked like. There were quite a few issues. Not being a network expert I don't know which of them are significant. The zip file has the wireshark capture file and both Studio log files for the same period. I hope there is enough information for you to see cause and effect.

# Test summary
Started packet logging
	Spurious retransmissions
	Dup acks
Started bambu studio
Went to device page - It took a while to actually display the printer - Select connect
	Errors with out of sequence packets - I think (this is the first time I have used wireshark)
After connected
Select camera play
Stop camera play
Load model, slice and select print
Failure to connect to printer with dialog to enter ip address and printer code
Select okay
Long pause and then printer is connected and details show
Select send
It goes to device page
Select camera play
I had forgotten to remove the previous print - firmware request submitted for confirm print removed
And the printer didn't care and started printing and created a spaghetti mess
Stopped printing from bambu studio
Stopped packet logging

Capture 
First packet:	2023-03-17 13:51:47
Last packet:	2023-03-17 14:05:43
Elapsed:	00:13:56
[P1P packet capture.zip](https://github.com/bambulab/BambuStudio/files/11006070/P1P.packet.capture.zip)

julie777, 2023-03-17:
It just hung. Here is the dump file.

I spoke too soon. The dump file is 1.2Gb and compressed it is 247Mb. Github only allows files up to 25Mb.

How can I send it to you?

bambu123, 2023-03-18:
Please wait for Monday, I think the discord may help.

bambu123, 2023-03-20:
@julie777 
Can you find me on Discord? My account is 'chunmao.guo#3806'.

julie777, 2023-03-20:
I am not very familiar with Discord. I sent a friend request to your account from julie777

bambu123, 2023-03-27:
@julie777 
I have checked the minidumps. It is the problem of live view. The d3d9.dll hung when started to create a window for video playing. 
It was a known issue but happened every time when playing the live view. That is the live view will never succeed to play which is not the same as your condition.
The old issue is not fixed yet because it is related to the system d3d9.dll. But we found that always happens when having dual graphics card in the system. A useful solution is to select an internal graphics card for BambuStudio in the Nvidia config panel.

From #909 
https://user-images.githubusercontent.com/2354924/210270636-77879795-106c-4e85-8cca-c50ff39f63e0.png

julie777, 2023-03-31:
@bambu123 Thanks for the update.

I looked at the screen capture of the Nvidia Control panel. 
Are you suggesting that I force Bambu Studio to use the integrated graphics instead of the NVidia graphics?

julie777, 2023-04-01:
I found the wiki article about laptops with discreet GPUs. 
I forced Bambu Studio to always use the built in GPU instead of the NVidia GPU. It has been 24 hours and I have not had a hang. I will keep you updated.

Also, the Wiki is outdated. It says to use the NVidia Control Panel. Windows 10 no longer works that way, but instead it controls GPU use in settings.

[image]

bambu123, 2024-02-06:
Fixed by user.

closed 2024-02-06 by bambu123