bambulab/BambuStudio #1362: Perimeter order

demek81, 2023-02-28:
Hi,

I'm wondering about perimeter order inner-outer-inner. I'll describe it on example. I have 4 perimeters, now this function forces to print it in this order: first it prints perimeters far from surface, than it prints outer perimeter and than it prints second perimeter. I think it should print 2nd first, than outer and than the others, because now it doesn't have any difference than outer to inner order. I mean outer perimeter can't stick to inner on overhangs. Or maybe i don't understand what this function is suppose to do so please elaborate. I think setting the order how i mentioned will help with adhesion, quality and accuracy. It will be compromise between inner to outer and outer to inner. But I'm not sure because i can't test it.

QingZhangBambu, 2023-03-01:
Shrinkage of the inner wall would affect the outer surface to cause the bed surface. The mode will improve the surface condition.
For example, if you print the 3rd and 4th perimeter first, then the 1st. Then the first wall cannot be pulled inward when printing the second wall, because the interior is already filled.
there is a pic to show that. The right cube was printed by the mode. left is outer to inner seq
[image]

demek81, 2023-03-01:
Are you sure left is outer to inner? Not inner to outer? If left is outer to inner they should look close to identical. I'll do some test prints, but as you said, printing outer perimeter on inner will affect surface quality. What I'm wondering is what  will happen when outer print is done on one inner perimeter not on 3 like in my example. Is printing more than 1 inner perimeters first will affect quality more?