bambulab/BambuStudio #1365: Allow Bridging overlap to override Wall Loops in Bridging area to some degree

ChaosBlades, 2023-03-01:
**Is your feature request related to a problem? Please describe.**
Bridging too close to outer parameter could end in thin air if Wall Loops are too high

**Describe the solution you'd like**
Allow Bridging Overlap to override Wall Loops to some degree only in the area where a bridge is located.

**Additional context**
With Wall Loops set to 4. Does not allow bridging extrusions to adhere to layer properly
[image]
Reducing Wall Loops to 2. Part is weaker overall but bridge has proper overlap.
[image]

QingZhangBambu, 2023-03-01:
Did you use the BambuStudio slicer? In BambuStudio, the bridge would be connected to ensure the bridge does not fail. 
as shown in the pic below.
[image]

if you use SoftFever's version or others, please ask them to fix it.

ChaosBlades, 2023-03-01:
@QingZhangBambu I am using the SoftFever Version but I tested the official build from BambuLab and I have the same exact issue. Also your picture is not what I am talking about.

Please re-open this issue. It is valid.

Here is the project file.
[Y Belt Anchor.zip](https://github.com/bambulab/BambuStudio/files/10856723/Y.Belt.Anchor.zip)

This is from official Bambu Studio. I am talking about the red arrow. That is not overlapping the wall loops. It is ending in thin air.
[image]

QingZhangBambu, 2023-03-01:
OK. I will check if the corner of the filament could ensure the print not be failed.

ncr, 2023-08-13:
Having the same issue - when I set walls to 3 or 4, my bridges are printed mid-air.

jkortus, 2025-06-11:
Happened to me when scaling down a model. When the wall gets too thin (related to more walls in the comment above), the bridges end in mid-air and collapse.

Example here: first larger model, scaled down and the last one is with "Bridge counterbore holes" option changed from "none" to anything else in Orca slicer (bambu studio does not have that option yet).

[image]

This is not a feature, it's a bug.
https://github.com/SoftFever/OrcaSlicer/issues/6264
https://github.com/prusa3d/PrusaSlicer/issues/9999 (prusa slicer has it resolved)

jkortus, 2025-06-11:
[bridge-debug-block.step.zip](https://github.com/user-attachments/files/20698312/bridge-debug-block.step.zip)
simple model to test, as the line gets thinner the bridges start to collapse