bambulab/BambuStudio #1366: Toolhead crashes on error when printing by object

SwagDad99, 2023-03-01:
**Describe the bug**
Printing by object, one object in each of the corners.  After the first two print, back right, and front right, an error happens when printing the back left object.  The toohead crashes into the front right object while trying to resume.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Print by object, with 'recover lost steps' ON in print options.
2. Print by object, an object in each corner of the bed, all the same height.
3. Tool head cover comes off while priting the left rear object.
4. When the cover is replaced, and 'Resume' is clicked, the tool head attempts to 'Home' in the front right of the bed, crashing into the objects already printed.
5. In order to resume printing, the two completed objects must be removed to prevent collisions with them.

**Expected behavior**
the bed should drop at least the amount of the maximum 'Z' height of the highest completed object before attempting to 'Home' the tool head.  THEN raise the bed back up to the previous height, and continue printing.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [Windows]
 - Version [11 Home]

SaltWei, 2023-03-02:
The resuming action is not generated by Studio. But acted by printer itself.
We will report this ticket to printer team

SwagDad99, 2023-03-02:
Thank you

From: SaltWei 
Sent: Wednesday, March 1, 2023 7:45 PM
To: bambulab/BambuStudio 
Cc: SwagDad99 ; Author 
Subject: Re: [bambulab/BambuStudio] Toolhead crashes on error when printing by object (Issue #1366)

The resuming action is not generated by Studio. But acted by printer itself.
We will report this ticket to printer team

TCC-up, 2024-01-19:
Hi, we have not reproduced the knocking over of other models on our side by following your steps. Can you submit a work order containing the error logs for this time-end and give us the work order number, thank you for your help in locating the suspected issue. You can also send us a 3mf file here first.