bambulab/BambuStudio #1372: Clean RTSP feed on LAN

user897943, 2023-03-01:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I want an unrestricted RTSP feed from the built-in webcam on my LAN at all times so I can add this to Home Assistant to monitor my prints. Currently it's possible to get some sensors and even control some things like the LED light via MQTT but the video stream is still not accessible unless Studio is running and only through a plugin for OBS. I want a normal stream URL directly from the printer, such as: rtsp://10.0.0.69/api/video/mainstream.m3u or such like. Then I can create a generic camera entity on Home Assistant or use whichever stream solution I choose. There's really no good reason not to do this over the LAN if the user chooses. Place a privacy option to toggle on or off in the printer firmware if there's any privacy concerns. The local stream should not affect the ability to do cloud streaming from Studio or Handy.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

bambu123, 2023-03-02:
VOD is not supported. We only push the stream to a UDP address where you can receive the stream if your software doesn't support RTP/UDP, please update it.
To configure the UDP address, edit the file 'ffmpeg.cfg' in the directory 'cameratool', then kill the bambusource.exe process and restart the 'Go Live'.

user897943, 2023-03-02:
Not sure if you understood.

I want an always-on rtsp/udp stream from the webcam. The printer is connected and has an IP, why can't you or won't you just give an endpoint for the webcam that's simple to use? Let's take a Hikvision IP camera, if I type this into VLC it will let me stream the video from the cam without modding anything: "	rtsp://192.168.10/Streaming/Channels/101" It's as simple as that. I don't want to modify anything and have to run your .exe file to spin up a stream, that is completely ridiculous.

On my £300 printer running Klipper (working behind me) I have a Logitech C920 and all I have to do is put this into VLC to get the MJPEG stream:
"http://192.168.1.10/webcam/?action=stream"

And my almost £2K printer the X1C with all the accessories and AMS can't do it? It's really ridiculous. I just want to be able to use a clean MJPEG or RTSP stream without fuss, so I can use it as an IP cam on any software.

bambu123, 2023-03-02:
OK, but this is a printer-side feature request.
I shall be helping transfer to the printer team.

sprior, 2023-03-02:
I agree that a video stream that does not require the Bambu Studio desktop to be involved is highly desirable.

sodachen22, 2023-03-02:
@user897943 @sprior Thanks for the suggestion. The video through RTSP is not encrypted, and they may cause some compatible problems for RTSPS. What do you think about the security issue?

user897943, 2023-03-02:
> @user897943 @sprior Thanks for the suggestion. The video through RTSP is not encrypted, and they may cause some compatible problems for RTSPS. What do you think about the security issue?

I don't care about encryption on the local network... It's a webcam inside a printer not a spy camera in the bedroom... Local IP access is enough for me. I can always use a reverse proxy if I want secure internet access (which I do). And a toggle to turn it on or off just like LAN mode would allow the user to choose if they want local unrestricted streaming via URL or encrypted cloud streaming the way it currently works.

sprior, 2023-03-02:
I agree, I don't personally care about encryption on the local network and don't require the stream to be accessible outside of my LAN.

sodachen22, 2023-03-02:
Thanks for the feedback. The requirement is recorded internally, we will define the priority and schedule based on feedback from more users.

gmartinez3, 2023-03-03:
Agreed, Enterprise grade Cameras on the local network routinely use RTSP with plain text credentials, it makes it easy to ingest into existing software.  A future enhancement could be to SSL Encrypt RTSP for public internet traversal, but that's not important for a first round.
I've had a request to ingest a Bambu stream into an Ava Video Management System and pure RTSP would do the trick.  It's not typical, but enables a whole world of integrations.

RyanEwen, 2023-03-03:
This would be amazing

sodachen22, 2023-03-03:
Which are the popular Home Assistant or Video Management System? Is there any other popular protocol besides RTSP suported by these systems? If there is, we can consider  to support a more secure protocol.

user897943, 2023-03-03:
Look, I totally understand as a business that security is extremely important and I know because of online tech blogs you are under a lot of pressure to secure everything but this is way overkill. Just put a switch on the printer settings to turn LAN streaming on/off and explain that it's unencrypted to allow integration into any software. Imagine if you were asking someone to encrypt their Hikvision camera or asking VLC to introduce encryption just for your implementation. It's not required. The cloud streaming is already encrypted, people can use that on the app. If you want, you can implement a limit on the number of connections, for example "allow X number of streams", so that if more than X number of connections are made it's not allowed past that number. Example: Allow 2 simultaneous streams. Then if a 3rd person tries to stream from the local URL on the LAN only, then they can't because the printer will deny it. But that's overthinking. This is a webcam inside a printer, gosh, let's not make a storm in a teacup, it's not a baby monitor inside a child's room. It's a camera in a printer and there's no microphone. The likelihood that someone will hack and use to spy is so low unless someone can prove it otherwise. We are not talking about a security camera here, let's not exaggerate the level of security required. A on/off toggle saying "stream over LAN (unencrypted)" is more than adequate. Keep the cloud streaming as-is so the slicer and phone app continue to work the way they currently do. The LAN stream should be in addition to what we already have, not as a replacement.

WolfwithSword, 2023-03-03:
+1 to opening up the RTSP camera stream direct from the printer on a lan network without needing bambu studio or the bambu cameratools as a middleman. This allows for any personal use-case to view the camera with a simple URL essentially, for example if a user is with a device without bambu studio or the handy app, and/or the app is not currently working. Additionally, we can then add the camera feed to our own personal monitoring systems.

gmartinez3, 2023-03-03:
@user897943 's point is exactly right.  There is not a more secure streaming method that is widely adopted, the highest grade Security Cameras stream with RTSP, and they're in schools.  This is a webcam in a printer, there's not danger of someone getting in, they'd need to already be on the local network and by that time you've got bigger problems.  Them simply seeing the stream or credentials, causes no lasting harm.
I'm the CTO for a Security company, and have to say I initially felt the same way, but the industry doesn't have an encrypted standard.  FFMPEG likely wouldn't even handle SSL encrypted RTSP, which is the only other step.  But you shrink integrations to less than 1%.
Take a look at streaming by Unifi, Verkada, Meraki, Ava, Avigilon, Milestone - they're RTSP.

RyanEwen, 2023-03-03:
RTSP and RTMP are where it's at. Disabled by-default (opt-in) is a fine compromise, if there needs to be one.

user897943, 2023-03-03:
Here's how to "secure" the unencrypted traffic: add a token to the end of the streaming url. Show the token on the printer LCD just like you are doing for the LAN mode authorization code. Let's assume as an example that the LCD on the printer had RTSP LAN streaming enabled by the on/off toggle, under that show a token such as "hkR47Fy4763gjEr". Assuming the LAN IP of the printer is 10.0.0.10 and the streaming URL is rtsp://10.0.0.10/cam/stream then to gain access the user would enter the following on their video software to access the RTSP on the LAN: rtsp://10.0.0.10/cam/stream?hkR47Fy4763gjEr and voila problem solved. Now only someone with direct access to the printer can get the token and form a valid URL. Using an incomplete URL or invalid token denies access to the stream. That's the "security" for LAN webcam access sorted. Send me a free X1C with AMS for the idea.

sprior, 2023-03-03:
Totally agreed with @user897943.  The only thing I'd ask while you're doing any of this is to provide a URL for a still image capture as well (this is extremely common on security cams) - not all usecases even need streaming video.

Kaloszer, 2023-03-03:
+1 for p1p

Crazyairborne, 2023-03-03:
+1 for RTSP stream. would be super useful for existing integrations

AdrianGarside, 2023-03-04:
+1 for this. It's my top desired feature by far.

djalexz, 2023-03-04:
+1 for RTSP stream

ltoolio, 2023-03-05:
+1 because of everything said before this reply.

PHIL-IP, 2023-03-07:
+1 for a rstp stream in local network

sodachen22, 2023-03-09:
Thanks for the feedback, an internal request bbl001-5087 is created, it will be evaluated and planned.

user897943, 2023-03-09:
Should be available in the next firmware update for sure.

cmeans, 2023-03-11:
+1 for a RSTP stream in local network.

maziggy, 2023-03-11:
+1 for a RSTP stream in local network.

Nerdiyde, 2023-03-12:
+1 for a RSTP stream in local network. :)

jumski, 2023-03-12:
+1 for RTSP stream!

jngmrks, 2023-03-12:
+1 for RTSP stream without need of a Cloud!

bastiedotorg, 2023-03-12:
+1 for a RSTP stream in local network.

Cobra-421, 2023-03-12:
+1 for RTSP stream without need of a Cloud!

uSyzex, 2023-03-13:
+1 for a RSTP stream in LAN.

greenhawk89, 2023-03-14:
+1 for RTSP stream accessible by Home Assistant or VLC. This is a must-resolve IMO and is pretty basic functionality.

scorfman, 2023-03-14:
+1 for RTSP. Thank you!

andreifinski, 2023-03-15:
+1
go rtsp, go!
I have to wire up my WYZE camera instead. people have already created mounts for it on thingverse.

felinh, 2023-03-15:
+1 for a RSTP stream in LAN.

LayerCakeMakes, 2023-03-15:
+1 would love to see this for home assistant integration, as well as camera support in bambu studio when not using the bambu cloud in lan mode.

Worren, 2023-03-16:
+1 for a RSTP stream

Flight777, 2023-03-16:
+1 Really a must have for Home Automation geeks :)

sprior, 2023-03-16:
For all the Home Automation geeks who want the clean RTSP feed, check out this suggestion to add more automation friendly MQTT topics: [https://forum.bambulab.com/t/recommendation-user-friendly-mqtt-options/2410/4?u=sprior](url)
I submit this as a feature on BambuStudio on github because this is a firmware feature not BS, but then I suppose neither is the clean RTSP feed.

user897943, 2023-03-16:
> For all the Home Automation geeks who want the clean RTSP feed, check out this suggestion to add more automation friendly MQTT topics: https://forum.bambulab.com/t/recommendation-user-friendly-mqtt-options/2410/4 I submit this as a feature on BambuStudio on github because this is a firmware feature not BS, but then I suppose neither is the clean RTSP feed.

The RTSP feed is absolutely firmware. I meant it must come from the camera and am confident everyone +1 this request feels the same. I don't wanna run a PC for a video feed. It's gonna cost a ton of money, it's inefficient to transcode when the printer board can do that all by itself using much less energy.

Definitely has to be on the printer firmware with no need to open the slicer or an app on a PC or mobile. Must be a straight URL from the printer.

user897943, 2023-03-16:
I suggest to also open the issue HERE on GitHub rather than the forum.

sprior, 2023-03-16:
OK, I just opened it here. I would think that a lot of the people interested in the RTSP feed would also want to support this request.  
https://github.com/bambulab/BambuStudio/issues/1504

Ali229, 2023-03-17:
+1 for a RSTP stream

frakki1986, 2023-03-17:
+1 for a RSTP stream

maniusman085, 2023-03-17:
+1 for a RSTP stream

FlyingHippo, 2023-03-18:
+1 for a RSTP stream in LAN.

ice-brainvoid, 2023-03-19:
+1 for a RSTP stream

HennieLP, 2023-03-19:
+1 for a RSTP stream in LAN.

MiezLisa, 2023-03-20:
+1 for a RSTP stream in local network (LAN).

franklinvv, 2023-03-20:
+1 For me, this is the biggest shortcoming of these printers.

RyanWor, 2023-03-21:
+1 for RTSP access

azure93, 2023-03-21:
+1 for RTSP. Need it !!

Photogad, 2023-03-21:
+1 for TRSP

jan1409, 2023-03-22:
+1 for RTSP access !!

KyleTeal, 2023-03-25:
+1 for RTSP

WolfwithSword, 2023-03-25:
Noticed the issue name changed to include an irrelevant issue, should be taken out and kept to just LAN video / RTSP stream. Firmware updating has several other duplicates and is unrelated to the camera stream, risks this issue being closed as a duplicate and lose progress on camera streaming.

user897943, 2023-03-25:
It won't be closed. This is a highly requested issue.

AdrianGarside, 2023-03-25:
You are just creating confusion and reducing the chance of either by lumping two unrelated issues together. That's now how issue tracking works.

They don't need each other. I am perfectly fine with connecting to bambu cloud whenever they release a new firmware update. That doesn't mean I don't want to be able to do that by sd card as well but it is wholly unrelated to a local video feed by by RTSP which is also wholly unrelated to LAN mode in the first place - I (and I suspect many people) want that even if not using LAN mode because I can then add it in other places (like my home assistant dashboard).

[19 more replies]

closed 2023-03-25 by user897943