bambulab/BambuStudio #1373: AMS color on printer screen not corresponding with color on Studio after syncing AMS

user897943, 2023-03-01:
**Describe the bug**
A clear and concise description of what the bug is.
I select what appears to be a gold color on the printer screen > AMS > slot 4 which contains a spool of golden colored filament (kinda mustardy in shade).

When I sync with studio, the color showing next to the 4th slow is orange, nothing like the one on the printer screen. There must be a better way of handling this. Use HEX or RGB so WYSIWYG in both devices, rather than some predefined presents. Better yet, let me sync colors from the PC to the printer as the PC color picker has unlimited possibilities. When I select the correct color on Studio and sync back, it reverts to the orange, and what's weird is that the shade on the printer screen doesn't look orange at all. Sync/Resync doesn't change anything. I'm using 1.5.0 public beta. Was the same on 1.4.2.x

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

bambu123, 2023-03-02:
Yes, we use RGB colorspace, which may display differently on both screens.
To check if it is a bug, please change the color to a different one, and see if it synced on both sides.

user897943, 2023-03-02:
It does not sync from the computer to the AMS, it only syncs from the AMS to the computer and the color is not the same. I get this sorta gold on the AMS (Generic PLA - three rows of colors, on the third row it's the one after black, white and grey (looks like gold/dark mustard) reading from right to left. When I sync with Studio, I get this clearly orange color. I noticed that the color list on the printer can be scrolled so why won't Bambu Lab just give more color options then. If you know that monitors have different ICC profiles and it's not consistent then why aren't you doing something to fix it? If you know the problem and you don't fix that means you don't care. But I care and want it consistent.

bambu123, 2023-03-02:
I mean there maybe a bug when sync RGB datas on the network.
If this is not a sync problem, then it must be a display diffirence.

Do you mean this third color?

[image]
From this one on printer
[image]

user897943, 2023-03-02:
No no, I mean the one next to grey!!! A few circles forward. The one before
grey, white and black.

bambu123, 2023-03-02:
Ok, this third one
[image]
From :
[image]

user897943, 2023-03-02:
The 2nd image does not appear... Looks like a failure upload your end.

I am talking about the 3rd row/line it's the 4th color from this way <-------- first is black, 2nd white, 3rd grey and 4th is the "gold" one. It looks a bit golden on my printer. And it shows completely orange on my PC. I am not color blind.

When I click the box on the PC I can pick unlimited colors. But it does not sync back to my printer! It works only from printer to PC and it is wrong color on the PC. Sync from PC to printer does not work for me!

bambu123, 2023-03-02:
Can you provide your screen snapshot? Is it different from mine?

user897943, 2023-03-02:
> Ok, this third one [image] From : [image]

That's the one, yes! On yours it looks consistent but try comparing on the actual LCD on the printer. The color doesn't look orange, it looks kinda like dark mustard which reflects on my filament. I know that the LCD on a printer won't be the most accurate thing but at least allow the customer to sync the color from the PC to the printer. That way, there is unlimited possibility to set the color and not just those 24. Currently Bambu Studio has unidirectional sync from printer to app, I want to be able to sync from app to printer, so that I have 16 million color choices instead of 24.

bambu123, 2023-03-03:
Having said that, what is your suggestion, to let the Bambu Hanny app provide more colors for selection?

user897943, 2023-03-03:
Yes and no. I want more color selection on any device. Bambu Handy, Bambu Studio Slicer and the Printer settings itself. I want to select any color I want for the filament. Better still if I can get HEX color and be able to manually input the color, that way I can take a photo of the filament, load into photoshop, get the HEX code for the color and put that exact color into the slicer, handy app or even on the printer screen and it will all be in sync.

MadMax389, 2023-03-04:
Can't you do that already from the AMS screen in Studio on PC? I know the sync is unidirectional from Device->Slicer, but I am able to select one of 16M colors on the AMS and then sync that exact color to the slicer. I do this all the time for different shades of the same color.
[image]

user897943, 2023-03-04:
If I pick the color from that exact screen on the slicer it does not sync
back to the printer. I switch my printer off when not in use to save
energy, so when it turns back on the AMS resyncs and the change is lost.
There's no reason why BL can't do bidirectional sync, they should. There's
no way to pick 16M colors from the printer, only the slicer... And if it
doesn't sync back it's pointless.

MadMax389, 2023-03-04:
Yeah, I understand. It makes it difficult if you are painting model but don't know yet which slots are going to contain which colors. You play around with the color palette in different slots and then lose them when you sync. For my workflow, I never select the AMS colors from the printer screen or the Handy app. For me, it would be a problem to sync the color palette from one particular slicer project to the AMS, since that would change the syncing to all my other projects.

user897943, 2023-03-04:
The AMS is supposed to display whichever color filament is in it... There is an assembly section on the 3mf file where you can put notes to keep track of your colors. But the AMS is supposed to always show the current color in each slot.

bambu123, 2023-03-09:
I think you change filament color from "Prepare" panel, not from AMS tray on "Device" panel, so the color is not saved to AMS.

bambu123, 2023-03-14:
OK, I think this is a feature request and not a BUG.
The colors with filament preset are not related to that in AMS slots even if they are synced.
It is designed so that we can build models without AMS and share models with others who may have not filaments with the same colors.
We suggest using the AMS Auto-Filament-Mapping to print the model, including models from the model store.