bambulab/BambuStudio #1376: Z-Offeset for P1P with textured PEI-Sheet is a hair to big

suit, 2023-03-02:
**Describe the bug**
I consistently get bed adhesion issues or 1st layer inconsistencies with all kinds of filament on both sides of the textured PEI sheet - the High Temperature Plate and the Engineering Plate don't have this issue.

The issue is especially worse when using Silk or Matte PLA from various vendors

**To Reproduce**
Steps to reproduce the behavior:
1. Slice your model for the Textured PEI Sheet and print it
2. see if the print sticks or produces a nice first layer

**Expected behavior**
The print should stick and have a consistent first layer

**Solution**
As mentioned here, chaning the Z-Offset slightly in the Machine Start code solves this issue: https://github.com/bambulab/BambuStudio/issues/941#issuecomment-1451729904

setting the Z-Value to -0.03 solves this issue.

i'm fully aware that this value is there to prevent the nozzle from scratching on the PEI sheet - but, since the nozzle probes the plate anyways, it will probe the high spots and then an additional offset is added, which makes the issue worse

since i only own one textured PEI sheet i can't evaluate if this solution is safe for all sheets - it needs to be evaluated if the value could reduced globally or at least can be insertet into per-sheet-variable that is used in the machine  - because when a user change this one value, it would exluded the entire machine start code from any further updates

adamfilip, 2023-03-12:
Needs to add a Z offset setting in the Filament profile.

suit, 2023-03-13:
> Needs to add a Z offset setting in the Filament profile.

in every filament profile? sure, why not, when it is an issue with the setting for a specific build plate that is more or less hardcoded in the machine start code ;)

GyroGearl00se, 2024-06-28:
On the Machine GCode I found following:

;===== for Textured PEI Plate , lower the nozzle as the nozzle was touching topmost of the texture when homing ==
;curr_bed_type={curr_bed_type}
{if curr_bed_type=="Textured PEI Plate"}
G29.1 Z{-0.04} ; for Textured PEI Plate
{endif}

I just duplicated the line "G29.1 Z{-0.04} ; for Textured PEI Plate"

(equals -0.08 ?)

And now the bed adhesion on my P1S with "Textured PEI Plate" is finally working again. I never had this issue in the beginning when I got my P1S, just started recently to have issues with Textured PEI Plate. No issues with "Smooth PEI Plate"