bambulab/BambuStudio #1381: X1C First layer Nozzle temperature not working

verityworks, 2023-03-02:
When Selecting a filament, setting the first layer temperature does not appear to have any effect while printing on the X1C.
It takes the Other Layer Temperature for the whole print, except the L shape calibration line.
(Similar to bug 1044 on the P1P)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Filament Settings
2. Change First Layer Nozzle Temperature to something other than the Other Layer Temperature
3. Slice a model, and look at the preview.
4. Select "Temperature" in the Color Scheme preview, to see what Temperature the GCode is supposed to be using

Expected to have the whole 1st layer set to this filament temperature

This bug appears to be fixed for the P1P machine, but I dont know how to get the GCode fix into the X1C settings.

 - OS: [Windows]
 - Version [1.4.x] AND [1.5.x] public beta

SaltWei, 2023-03-03:
[image]
Seem to be OK for X1C

SaltWei, 2023-03-03:
The calibration line should be printed by other layer temperature to make the PA calibration match with most part of model

verityworks, 2023-03-03:
Diving deeper, it seems ok with just a simple STL that doesnt require support.
I;m trying to print PETG with a PLA interface layer as support, and then I get this as the first, and all other layers.
[image]

SaltWei, 2023-03-07:
We can reproduce this issue now.
[image]
This issue is caused by change filament gcode, and after switching filament in first layer, the temperature is always using other layer temperature without judging whether is in first layer or not.

We will fix it, thank you very much!

zhimin-zeng-bambulab, 2023-04-18:
It fixed in v1.6 released on github.

closed 2023-04-18 by zhimin-zeng-bambulab

joshthehappy, 2024-07-01:
I'm seeing something like this right now, I set a first layer temp and the printer or the slicer is ignoring it. Weird thing is it used to work just check the box and good to go.