bambulab/BambuStudio #1382: AMS feature rqst- function to easily alternate colours in selected pattern at selected distance (make your own rainbow/gradient print combos from standard colours)

3DMonkeyAUS, 2023-03-02:
**Is your feature request related to a problem? Please describe.**
AMS feature rqst- ability to use AMS to alternate colours in selected pattern at selected distance (make your own rainbow/gradient print combos)

isnt it annoying when you cant find the perfect colour combo in rainbow or gradient filaments, and when you do, dependant on the size of your model you the filament may change either too fast or too slow which means you either see to much or too little colour change.

**Describe the solution you'd like**
if the AMS was able to support the ability to print specific lengths of filament from each connected/selected spool in whichever whichever pattern/sequence the user selected, this would mean users could have infinite possibilities for colour combos and visual effects.

**Describe alternatives you've considered**
alternatives are lame, go with the cool easy to use option ;)

lanewei120, 2023-03-07:
Could you show some pictures of the effects you want to print?

3DMonkeyAUS, 2023-03-07:
> Could you show some pictures of the effects you want to print?

[image]

so rather than using a rainbow filament, you can use your own chosen colours and adjust how frequently they change to create the desired appearance for each print.

some filaments change colours to fast for smaller prints or too quick for larger prints.