bambulab/BambuStudio #1386: AMS Filament Backup missing in v1.5 public beta on P1P

Hologos, 2023-03-02:
**Describe the bug**
Even though release notes mention that P1P now supports AMS Filament Backup Feature, I am not able to find in the slicer nor in Bambu Handy (all latest firmware (slicer, mobile app, ams, p1p)).

If I click on AMS settings icon in the slicer, I get an error "Current AMS settings does not support using an external spool."

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: macOS 13.3

Hologos, 2023-03-02:
Once I added the same color to a slot, I was able to open AMS settings in the Slicer on desktop. But Bambu Handy still doesn't show this option.

[image]

StoneLiBambulab, 2023-04-03:
Hi @Hologos ，
    Select any AMS slot to view the ams setting. If the ext. spool is selected, then cliking the ams setting icon, it will pop up this error message. The AMS auto switch filament is the same option as AMS filament backup, i will refine the description. And the feature will be supported in recent printer firmware version, please wait for the printer firmware and upgrade it.

Thanks!

Stone Li

Hologos, 2023-04-03:
> Select any AMS slot to view the ams setting. If the ext. spool is selected, then cliking the ams setting icon, it will pop up this error message.

I managed to fix that as mentioned in 2nd comment. It would be great if the AMS setting could be opened regardless of selected spool. Or at least to get error message saying the reason.

> The AMS auto switch filament is the same option as AMS filament backup, i will refine the description. And the feature will be supported in recent printer firmware version, please wait for the printer firmware and upgrade it.

As mention, I was on latest firmware on all devices (printer, Bambu Studio and Bambu Handy). The AMS filament backup feature was added in the latest release of Bambu Handy which was released 3 days ago.

StoneLiBambulab, 2023-04-03:
Hi @Hologos,
    ok. the selection logic is designed for multi AMSs, and it can be optimized if only one AMS exists.  i will refine the logic.

Thanks!

Stone Li

Hologos, 2023-06-26:
@StoneLiBambulab This can be closed I guess.

closed 2023-06-26 by Hologos