bambulab/BambuStudio #1387: Add Pause icons are difficult to see due to size and color

moop2000, 2023-03-02:
"Add Pause" icons are difficult to see due to size and color in the preview screen.

The icons need to be larger, and different in color so as to easily stand out against the background color.

**Screenshots**
Example of icons in question.
[image]

Example of difficulty seeing icon on a full screen. It is half way up the layer indicator.
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11 21H2
 - Screen: 3840x2400, Scale: 200

LiZ-Li-BBL, 2023-03-03:
Thanks for your suggestion, a new icon design will be applied in the future.

LiZ-Li-BBL, 2023-03-31:
We noticed that the scroll bar and thumbnail in your preview page are not in the right scale. This is because your screen resolution is too high. You can temporarily turn the screen scale down to 100% to get a normal size of preview items. We will add the adaptation for high resolution screen in the near future.

LiZ-Li-BBL, 2024-01-16:
We added a more visible mark beside the tick.

closed 2024-01-16 by LiZ-Li-BBL