bambulab/BambuStudio #1389: Pausing immediately after first layer inspection to manually clean nozzle resulted in no resume button

eisch067, 2023-03-03:
**Describe the bug**
I pressed pause following first layer inspection to clean the nozzle tip manually due to the nozzle wiper missing a blob on it. Cleaned it and went to hit resume, and the screen only showed pause. Bambu Studio also only showed pause. Pressing pause did nothing. Had to stop print and start over. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
Should have just resumed like it always does. 

**Screenshots**
https://user-images.githubusercontent.com/60443782/222624801-9a825b22-5739-46ef-aead-84ddf12a066d.MOV

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

SaltWei, 2023-03-03:
Not slicer issue. @sodachen22 Please have a look

SaltWei, 2023-03-03:
By the way, more and more user come to Studio github to report printer issue， where is not the right place.
How about create a Printer project for user to report issues, just like https://github.com/bambulab/Bambu-Handy/issues
@sodachen22