bambulab/BambuStudio #1391: Dark mode on Ubuntu

dovebman, 2023-03-03:
When I use dark mode on my linux system, some textbox do not apply the white text color.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Ubuntu Settings
2. Click on Appearance
3. Click on Dark
4. Open the Bambu Studio
5. Click on Printer settings
6. Go to Machine G-code
7. See the bug

**Expected behavior**
Dark textbox with white texts

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 22.04.2 LTS

TexZeTech, 2023-03-22:
I have this in Manjaro as well 

**Version**
v1.5.0 Hotfix1 Ubuntu.appimage

**Desktop & hardware**
Operating System: Manjaro Linux
KDE Plasma Version: 5.26.5
KDE Frameworks Version: 5.103.0
Qt Version: 5.15.8
Kernel Version: 5.15.102-1-MANJARO (64-bit)
Graphics Platform: X11
Processors: 8 × Intel® Core™ i7-3770K CPU @ 3.50GHz
Memory: 15.6 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 1080/PCIe/SSE2

walterwongbbl, 2023-03-23:
Hi.
This is a bug and that we are fixing.

walterwongbbl, 2023-05-25:
Hi.
The next version will fix this issue.

closed 2023-10-13 by walterwongbbl