bambulab/BambuStudio #1392: V1.5.0 Public Beta closes on Mac

ScreamXY, 2023-03-03:
**Describe the bug**
I open Bambu Studio and prepare a model. Save the .3mf file and walk away to get a coffee. When I come back to my Mac Bambu Studio is closed.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Bambu Studio V1.5.0
2. Add .stl file, slice and save .3mf
3. ctrl + cmd + q to lock
4. Wait till the machine goes to sleep
5. Log back in -> Bambu Studio is closed

**Expected behavior**
Bambu Studio stays open.

**Desktop (please complete the following information):**
 - OS: macOS Ventura 13.2.1
 - Chip: M1 Max
 - Bambu Studio: 1.5.0.59

SaltWei, 2023-03-06:
> When I come back to my Mac Bambu Studio is closed.

Could you share the log files under Help->show configuration folder ?

ScreamXY, 2023-03-07:
Here are my logs of today, where it happened.

[debug_Tue_Mar_07_19_35_32_15134.log](https://github.com/bambulab/BambuStudio/files/10912890/debug_Tue_Mar_07_19_35_32_15134.log)
[debug_Tue_Mar_07_16_12_04_316.log](https://github.com/bambulab/BambuStudio/files/10912891/debug_Tue_Mar_07_16_12_04_316.log)
[debug_network_Tue_Mar_07_19_35_32.log](https://github.com/bambulab/BambuStudio/files/10912892/debug_network_Tue_Mar_07_19_35_32.log)
[debug_network_Tue_Mar_07_16_12_04.log](https://github.com/bambulab/BambuStudio/files/10912893/debug_network_Tue_Mar_07_16_12_04.log)

lanewei120, 2023-03-08:
we also tried on our mac laptops(m1 based, os 12.6, 13.2.1),
and we can not reproduce it currently

and from the log above, we can not identify the root causes.

could you also help to find the crash report in the "console" application
and upload here?
@ScreamXY @Hologos

ScreamXY, 2023-03-08:
@lanewei120 could you guide me on how I can get that crash report?

lanewei120, 2023-03-08:
just found it in the "Console" app

https://support.apple.com/en-gb/guide/console/cnsl664be99a/mac

ScreamXY, 2023-03-09:
That's the problem, it does usually not crash. It's just closed. When started it also asks to reload certain things.
I got this crash report from the very first day of using this version. But since then I have no logs but still the behaviour.
[BambuStudio-2023-03-03-154824.ips.zip](https://github.com/bambulab/BambuStudio/files/10928644/BambuStudio-2023-03-03-154824.ips.zip)

bambu123, 2023-03-10:
Are there more crash reports about BambuStudio in your Console App?
This one shows the app was exiting and crashed with the About dialog showing.
Can it be closed when your macOS user session logout.

ScreamXY, 2023-03-10:
There are no other crash logs.
I do not log out of my session. I just lock the computer and come back later.
It seams that it does not happen to be closed when there are multiple windows of the studio.
I feel it happens when I have a project open and change something basic like seam position and infill. Like there have to be changes. But I couldn't reproduce yet.
I updated to the new public release last evening. Will report back if the issue still occurs.

Hologos, 2023-04-15:
Hi, here is a crash log + the 3mf project.
[bambu-studio-crash.zip](https://github.com/bambulab/BambuStudio/files/11239212/bambu-studio-crash.zip)

bambu123, 2023-04-18:
@Hologos The crash is another issue relate to the SSDP stack.

closed 2026-10-13 by ScreamXY