bambulab/BambuStudio #1393: [feature request] Paint mouse/brim ears aka anti warping tab

guillaumemolter, 2023-03-03:
**Is your feature request related to a problem? Please describe.**
Brims, waste a lot of filaments and are annoying to cleanup. Mouse ears are very commonly viewed as a better alternative to brims. Right now to add mouse ears you need to either [import some additional STL](https://www.printables.com/model/282822-mouse-ears-dissected/) and spend a lot of time placing them, or by adding a primitive cylinder and a negative cube to achieve a proper mouse ear without messing up the walls of the actual model.

[image]

**Describe the solution you'd like**
Similar to support painting, I want a mouse ears painting tool that allows me to place mouse ears with one click. This is the plugin I was using with Cura and does exactly what I want: https://github.com/5axes/TabAntiWarping

**Describe alternatives you've considered**
Alternatively you could add a right click option "add brim ear" that would drop the correct shape on the bed and let you place it. Really not ideal but better than nothing.

**Additional context**
https://www.youtube.com/watch?v=MCcFMDv_4eo

mlhilljr, 2023-03-24:
I would like to second this request. The Brim Ears in Super Slicer is a feature I use on every print. The ease of use and removal is simply amazing!

Fredyy90, 2023-03-28:
Please add this feature soon, I used it all the time in Cura, and now I really got issues printing some of my models on my x1c. As I don't want to go to a full brim

Fredyy90, 2023-04-15:
Any Updates on this feature?

Fredyy90, 2023-04-28:
@StoneLiBambulab did you already discuss this? I could use this feature every day, right now, and avoid countless failed prints.

Btamasb, 2023-05-01:
We wait for the "mouse ear" function, there is some news from him?

StoneLiBambulab, 2023-05-10:
I'm sorry. We have some higher priority issues to deal with after discussion for next version (1.7)

Thanks!

Stone Li （旺仔）

Calzune, 2023-06-28:
Big +1 on mouse ear funktion

gitagogaming, 2023-07-10:
+1 for mouse ears...

niklasdevries, 2023-07-14:
+1 for this!

CumpsD, 2023-07-17:
No idea if this is the same feature, but it would be nice to draw on brim locations. I have parts of my model which really need a brim for the small details, but the rest of the model doesn't need it, and even suffers from a brim.

josefdahari, 2023-07-31:
Funny I was just looking for this option or feature in Studio and realized it's not available +1 for this as well.  Can come in handy for sure !

Thank you

mendylivium, 2023-08-11:
I really need this.

voglster, 2023-08-17:
Just adding my +1

DavidCaliri, 2023-08-20:
This is the only reason why it's so difficult to leave SuperSlicer.  I use this feature on every print with sharp corners!

SaltWei, 2023-08-21:
We have add initial version of brim ear in https://github.com/bambulab/BambuStudio/releases/tag/v01.07.04.52 by using feature 2 and feature 3 together. One important thing is that brim ear can use different filament for example support for PLA.
Similar feature request in https://github.com/bambulab/BambuStudio/issues/689

Of course, this is not the best interactive mode. We will try to allow user to use mouse left click to pick which corner needs brim ear or pick corner by angle threshold(may not accurate when model is complex). 

We will keep improving this function.

the-robo, 2023-08-23:
Please keep working on this feature!

tomparvin, 2023-08-24:
Would also like to see this feature added! Big time!

visualdensity, 2023-09-01:
Me too!! Would deeply appreciate being able to paint mouse ears or brim. Thank you!

IanAlbinson, 2023-09-24:
I'd love to see "Mouse ear" brim type ported over from Orca Slicer 1.7.0. It's so incredibly useful!

KeithinPHX, 2023-10-23:
Mouse ears please!

fangly, 2023-10-25:
[V1.7.6](https://github.com/bambulab/BambuStudio/releases/tag/v01.07.06.92) (feature 3) introduced the disc primitive, which serves as a poor man's brim ear.
Looking forward to future developments.

Geekmommy4, 2023-10-25:
> [V1.7.6](https://github.com/bambulab/BambuStudio/releases/tag/v01.07.06.92) (feature 3) introduced the disc primitive, which serves as a poor man's brim ear. Looking forward to future developments.

Could you get that to work? I couldn't get it to print the disc when I sliced it.

Fredyy90, 2023-10-25:
> Could you get that to work? I couldn't get it to print the disc when I sliced it.

It's possible to get the disc printed, but it'll always merge the discs into the models, so the outer walls are not nice and clean as they are with a brim or as they should be with brim ears.

I always get them printed like on the left, when it should be like on the right.
[image]

Geekmommy4, 2023-10-25:
> It's possible to get the disc printed, but it'll always merge the discs into the models, so the outer walls are not nice and clean as they are with a brim or as they should be with brim ears.
>
> I [...]

Makes sense...thanks!

PhilBaz, 2023-12-20:
Yes, this is a critical feature that is missing. One of the best features of Cura. Its called tab anti-warping. Its very important for us ABS users.

SR-G, 2024-03-11:
Any progress on this ?
Regarding my personal needs, this is one of the top-most missing features at this time in Bamby Studio ...

KiLLie2k, 2024-03-26:
+1 back in cura for tab anti warping for a specific piece of a model, would love to stay in orca as this specific printer is well configured for it.

lorenzofanchi, 2024-06-18:
+1 would be amazing to see brim ears soon!!

tauceti82, 2024-06-24:
+1 I also highly need this! Don't want don't want to use orca slicer

SaltWei, 2024-06-24:
We are working on it at the moment.
Not just copy from superslicer, but different function which can paint brim ear.
Highest priority for next big version.

CraigHutchinson, 2024-09-12:
If we could add this as a `Modifier` volume would be amazing as a UX feature to solve all/any use-case imho. Adding a Cyl;inder to turn on Brim where needed or for parts of a model would be perfect.

TheRealHotFix, 2024-10-05:
+1 for this request. I almost installed Orca Slicer several times just to get this feature, but I keep coming back to Bambu Studio.

IanAlbinson, 2024-10-05:
@Dan Just instal it. It’s great and the mouse ears are amazing.

MackBambu, 2024-10-18:
https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.74
Brim ear has been added.

closed 2024-10-18 by MackBambu

PhilBaz, 2024-10-18:
@MackBambu Amazing! Thank you!

Geekmommy4, 2024-10-18:
Thank you so much!!

TheRealHotFix, 2024-10-19:
> https://github.com/bambulab/BambuStudio/releases/tag/v01.10.00.74 Brim ear has been added.

Thank you VERY VERY VERY much!