bambulab/BambuStudio #1394: Solid Infill Pattern option is missing

fredenigma, 2023-03-03:
At the moment it is possible to choose infill pattern for top and bottom surfaces as well as for the sparse infill. 
for the other layers filled with a solid infill i.e. if you set to have 6 top layers, the 5 below the topmost surface, the slicer automatically sets the rectilinear infill and there is no way to switch to another pattern.

It can be useful to choose different patterns for solid layers infill, also because the strength of the parts may vary when changing the infill pattern of the outer shell (entirely, not only the cosmetic inside and outside).

It is an option explicitely present in PrusaSlicer, so it should be very easy to port it to BambuStudio, only a "Solid Infill Pattern" item in the slicing option would be required.

SaltWei, 2023-03-07:
> It is an option explicitely present in PrusaSlicer

Could you figure out which one is in PrusaSlicer？
[image]

fredenigma, 2023-03-07:
[image]
It was in Super Slicer (First line, solid) not PrusaSlicer, sorry.

SaltWei, 2023-06-20:
same with https://github.com/bambulab/BambuStudio/issues/1019

QingZhangBambu, 2023-08-01:
we've added that on version 1.7.  You can update your software to get that.
[image]

closed 2023-08-01 by QingZhangBambu