bambulab/BambuStudio #1395: New P1P firmware broke mqtt connection unless lan mode is turned on

AdrianGarside, 2023-03-04:
The X1 did not require lan mode when the firmware update to enable secure mqtt was made. But the P1P firmware does. This broke my ability to connect my bambu to my home automation system. Please fix the secure mqtt connection to not require lan mode.

sodachen22, 2023-03-07:
@AdrianGarside Thank you for the feedback. Because of the limitation of CPU and memory resource in P1P, local MQTT server is enabled only in LAN mode. We will optimize the system and check if MQTT server can be always on.

AdrianGarside, 2023-03-07:
For what it’s worth, I was running with it permanently feeding into home assistant for the last couple of weeks with no noticeable negative impact on printer behavior.

andreifinski, 2023-03-07:
for me even the LAN mode is not working. also there is something to do with the self-signed certificate in place. please sign it with some trusted advisory

AdrianGarside, 2023-03-08:
> for me even the LAN mode is not working. also there is something to do with the self-signed certificate in place. please sign it with some trusted advisory

That's a different issue. I see that also with mqtt explorer and it blocking for a successful connection. But the home assistant integration is configured to be more accepting so isn't blocking on the self signed certificate.

totalitarian, 2023-03-24:
Hi, any update? Really missing the mqtt

LayerCakeMakes, 2023-03-28:
+1 please unbreak mqtt

byWulf, 2023-03-28:
+1 Please give us access to the p1p mqtt. I don't want to enable LAN only mode because I still want to get automatic updates without any hassle. Thank you guys!

dadonali, 2023-03-28:
+1 in the first Firmware Version it works too without using LAN Only Mode 🙁

mike391, 2023-03-29:
+1 MQTT and ftp should work regardless if in lan mode or not. 

The only thing lan mode should be doing is connecting/disconnecting from the cloud.

byWulf, 2023-03-30:
An alternative for the missing MQTT without LAN Only Mode could also be giving us access to the cloud MQTT from us.mqtt.bambulab.com

That way you don't have the mentioned problem with the limited CPU and memory resources, because of my understanding this MQTT has always the needed data as it is the source for the BambuLab Studio status and BambuLab Handy.

What do you think about that?

mike391, 2023-03-30:
> An alternative for the missing MQTT without LAN Only Mode could also be giving us access to the cloud MQTT from us.mqtt.bambulab.com
>
> That way you don't have the mentioned problem with the limited [...]

That's an alternative that's better than nothing. but not a good one, Bambu cloud can go down (and has) plus alot of printers are not in an environment with internet access. Local access is optimal. Resources isn't an issue because the mqtt and ftp servers are still running even though we aren't able to access them.

DzzD, 2023-04-01:
+1 (specially for home automation without internet)

RyanEwen, 2023-04-02:
+1 - would really like to use the Home Assistant integration without using LAN mode

RyanEwen, 2023-04-06:
Seems like the latest P1P firmware from a few days ago allows for MQTT while outside of LAN mode, but I don’t see the topic or anything published to it once connected

AdrianGarside, 2023-04-07:
I'm able to connect again now even when in cloud mode.

totalitarian, 2023-04-07:
> I'm able to connect again now even when in cloud mode.

I can connect but there is no topic?

closed 2023-04-29 by AdrianGarside

DzzD, 2023-04-30:
Anyone able to connect using mosquitto_sub ?

krokus57, 2023-08-31:
Also need camera lan liveview like in X1C [https://www.wolfwithsword.com/bambulabs-x1c-camera-in-home-assistant/](url)