bambulab/BambuStudio #1396: Missing Prime Tower for Filament Change from Preview Tab

EukaDesigns, 2023-03-04:
**Describe the bug**
When changing the filament at a specific layer height by right clicking the layer after slicing, and then re-slicing, no prime tower is generated. When painting, the prime tower is generated. 

**3mf File for This Bug**
https://1drv.ms/u/s!Atw9blB-jp7OgolLxpCmkmpMQRge0g?e=NnQKuo

**To Reproduce**
Steps to reproduce the behavior:
1. Slice file
2. In preview tab, drag '+' layer indicator to layer for desired filament change. 
3. Right click '+', hover over change filament and then click desired filament. 
4. Click slice again
5. Observe that no prime tower is generated. 

**Expected behavior**
A prime tower should be generated. 

**Desktop (please complete the following information):**
 - OS: Windows
 - Windows Version: 11
 - Bambu Studio Version: 1.4.2.13

SaltWei, 2023-03-06:
@EukaDesigns After inserting filament change at certain height, please go to prepare first and then re-slice. Then the wipe tower can be generated.
@zhimin-zeng-bambulab Please check and fix this bug.

zhimin-zeng-bambulab, 2023-03-06:
It is a bug and we will fix it in the next version.

EukaDesigns, 2023-03-12:
I just downloaded v01.05.00.60 and the prime tower is generated when filament changes are made in a layer in the slice window. Thanks!

closed 2023-03-12 by EukaDesigns