bambulab/BambuStudio #1397: [Support interfaces missing when slicing object.] Support interfaces missing or not generated

BambuLabDiscord, 2023-03-04:
Support interfaces missing when slicing object. with the option for support only the build plate. This occurs when you slice a hollow object that have platform inside it.
Even on the purge tower is smaller than needed.

MusangKingBambulab, 2023-03-06:
Hi @BambuLabDiscord, would you mind sharing a specific 3mf with me? That would be convenient and effective to reproduce and locate the bug for me. Thx.

davidpyo, 2023-04-17:
[StockShell.zip](https://github.com/bambulab/BambuStudio/files/11254143/StockShell.zip)

Selecting "On Build Plate Only" causes support interface to not be generated : 
[image]

Unselecting "On Build Plate Only" causes support interfaces to be generated correctly: 
[image]

MusangKingBambulab, 2023-04-19:
Hi, @davidpyo, I have locked the bug and am busy fixing it. Actually, for this model, they can be printed without support. You can click Support->Advanced->Don't support bridges and will find this model can be printed in support-free. As for the bug, I will solve it in the next release. Thanks!
[image]