bambulab/BambuStudio #1398: Retain Last Setting In Text Tool Dialog

MadMax389, 2023-03-04:
**Is your feature request related to a problem? Please describe.**
The new text tool in v1.5 is very useful, however the font, size, and embed depth are not retained when adding new text. In version 1.4, when new text was added, it retained the settings from the previous application of text. Now for every application of text, you must re-enter all the pertinent information.
**Describe the solution you'd like**
Maintain previous text settings in the dialog box so it is not necessary to re-enter the settings e.g. when adding multiple lines of text using the same settings for each line.

**Describe alternatives you've considered**
Text settings applied..

[image]

Test settings revert to default. It would be nice to maintain the previous setting between application of the text tool (like v 1.4 did)
[image]

**Additional context**
BS Version 1.50.00.59
Windows 11

StoneLiBambulab, 2023-03-10:
Hi @MadMax389 ,
Thank you for your good advice, we will consider optimizing in the recent version.

Stone Li

Haidiye00, 2026-09-22:
https://github.com/bambulab/BambuStudio/releases/tag/v02.08.04.57
Hello, please update to the latest version. The latest version is already supported. Thank you for your support. If you have any further questions, please feel free to raise them. We will continue to communicate and thank you again

closed 2026-09-22 by Haidiye00