bambulab/BambuStudio #1402: Ability to Minimize the Settings panel to get view to all parts

Benderrodrigu3z, 2023-03-04:
Scrolling down to locate the part can be a challenge, sometimes the scroll bar jumps and you lose what part was the part you were working with after changing the type to a negative part, can we get the ability to minimize the bottom section, perhaps titling this section as Part Options so it has a header for minimizing

[image]

lanewei120, 2023-03-07:
[image]

you can click at the blank row below the "outside"
then the "part options" will disappear

Benderrodrigu3z, 2023-03-07:
No right when you click on an object it comes back, I mean a permanent hide like the other options haveSincerely,Shane FSent from my iPhone13