bambulab/BambuStudio #1403: A4 C3 not Everywhere implemented

Benderrodrigu3z, 2023-03-04:
Please Unify my original idea on making the A1-4 B1-4 C1-4 D1-4 idea everywhere in the Slicer when im Missing a roll these values don't match

[image]
[image]

StoneLiBambulab, 2023-03-10:
Hi @Benderrodrigu3z ,
Here is a list of filaments used in a print project, which is different from the concept of the filaments in AMS of a physical printer.

Thanks!

Stone Li

Benderrodrigu3z, 2023-03-10:
So you’re telling me that the BambuStudio app has to SUFFER because it supports other printers? Like what the hell?

First off I can tell BambuStudios what printer I have so where is the issue in doing A2 B3 ?