bambulab/BambuStudio #1405: Seam gap & acceleration controls

npoulos, 2023-03-05:
**Describe the bug**
Can you please port over the seam gap enhancement from softfever. I reported this previously and the developer said it showed no difference in the seam when changing to 0 but it actually does and many people have found it useful. It's such a simple fix and already done in softfever so just because your one developer couldnt find use for it doesnt mean it's not useful.

The seam gap also works really well with the ability to limit the acceleration of travel moves so please also port over the extra acceleration controls. 

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
macOS

npoulos, 2023-03-05:
https://github.com/SoftFever/BambuStudio-SoftFever/issues/166

npoulos, 2023-03-05:
https://github.com/bambulab/BambuStudio/issues/1063 issue where dev decided not to implement without discussing it further.

QingZhangBambu, 2023-04-21:
#1247

QingZhangBambu, 2023-04-23:
What machine did you use? i test that in Bambu X1C. The result is no difference

npoulos, 2023-04-23:
Tested on both X1C and P1P using 0.2 nozzle. 

You may not be testing with properly tuned K value.

QingZhangBambu, 2023-04-23:
I will try that with 0.2 nozzle. I used 0.4 nozzles before.

QingZhangBambu, 2023-04-23:
Plus, there is the test result of 0.4 nozzle
[image]

QingZhangBambu, 2023-04-24:
There is the result of 0.2 nozzles. PS: I used Orca and apply the parameters you provided. The result still not good
[image]

QingZhangBambu, 2023-04-24:
Could you provide your result and 3mf files?

QingZhangBambu, 2023-08-08:
we open "seam gap" for develop mode, hope that would be helpful.

closed 2023-10-16 by QingZhangBambu