bambulab/BambuStudio #1406: Incorrect link in Tip of the Day: Stacking objects

Hologos, 2023-03-05:
**Describe the bug**
Link leads to edit page of Stacking Objects `https://wiki.bambulab.com/e/en/software/bambu-studio/stacking-objects` instead of `https://wiki.bambulab.com/en/software/bambu-studio/stacking-objects`.

**Screenshots**

[image]

[image]

[image]

**Desktop (please complete the following information):**
 - OS: macOS 13.2
 - Version: Bambu Studio 1.5.0.59

LiZ-Li-BBL, 2023-03-06:
Yes, that's a mistake. It has been fixed and will work normal in the next version.

LiZ-Li-BBL, 2023-04-21:
Fixed

closed 2023-04-21 by LiZ-Li-BBL