bambulab/BambuStudio #1407: Change settings per color

ChrMaass, 2023-03-05:
**Is your feature request related to a problem? Please describe.**
When I do multi-color printing with both solid and translucent filament, I always have to choose the optimal setting for the translucent filament. I cannot optimize the print according to the corresponding filaments in terms of infill, waste in the object etc.

**Describe the solution you'd like**
It would be great if you could change (some) settings for different colors of the same object. For example, for solid filaments you could select the option that the flush may be performed in the infill (and not in the translucent filament) and that the translucent filament, for example, is printed solid, while the other only receives 15% infill.

suit, 2023-03-05:
You already do this, if you import it as a multiple parts and assign each part different settings

ChrMaass, 2023-03-05:
Yes, exact like this but for non multipart objects. It could be shown like a multipart. Or an option to split the object in a multipart object by color.

SaltWei, 2023-03-06:
> Yes, exact like this but for non multipart objects. It could be shown like a multipart. Or an option to split the object in a multipart object by color.

Good idea!