bambulab/BambuStudio #1411: slight layer shifts on manual filament change (with pause)

suit, 2023-03-05:
**Describe the bug**
when the filament is changed manually during print, there is a slight "random" layer shift (x/y), that is not a result of the gcode itself

this offset appears do not have something to do with the gcode itself - see attached code, the pause and filament change is on L25 and L26, the X/Y values are exactly the same for all the layers - so the resulting gcode is in order

i assume that the filament change, nozzle wiping and moving around the toolhead screws up the homing information by a hair and thus causing layer shifting

i've printed multiple parts so far, the layer shift seems to always lean to the rear/left side of the printer - it ranges from about 0,1 to 0,5 mm

i tried this with a few of those test 10 mm³ cubes and i always get a shift - i've also been very carful to note move, touch, or shake the printer in any way - if you use the same gcode and don't change the filament and resume right away, you don't get a shift

**3mf File for This Bug**
[pause+filament change_next_layer.gcode.gcode.txt](https://github.com/bambulab/BambuStudio/files/10892007/pause%2Bfilament.change_next_layer.gcode.gcode.txt)

**To Reproduce**
Steps to reproduce the behavior:
1. invoke pause at a certain layer
2. change filament manuallyö
3. resume printing
4. see, if the layers are aligned

**Expected behavior**
X/Y position stay exactly the same on resume

**Screenshots**
[image]
left side is also warped due to another issue already reported, but the layer shift is also pretty substantial here

Hologos, 2023-03-05:
How long did it take for you to change the filament? The motors get powered off after 15 minutes of idle time, that's when the shift happens. It should not happen if you make it in this time and handle the toolhead delicately.

SaltWei, 2023-03-06:
Not caused by slicer and gcode.
I have transfered this issue to printer team.

suit, 2023-03-06:
> How long did it take for you to change the filament? The motors get powered off after 15 minutes of idle time, that's when the shift happens. It should not happen if you make it in this time and handle the toolhead delicately.

i printed those little cubes for testing and i stood right next to the printer, as soon as the printer paused

sodachen22, 2023-03-07:
@suit It looks there is problem with your printer, could you please create a ticket with related LOG?
https://wiki.bambulab.com/en/x1/troubleshooting/upload-log

suit, 2023-03-07:
@sodachen22 since i've printed dozens of this cubes, i can't find it in the log right now - but i'll reprint them later this week and create a ticket, there

Hologos, 2023-03-07:
I can confirm that if I disconnect AMS and try to manually change the filament, I get a layers shift as well. I also noticed that if I ran out of filament and there is not any suitable filament in the AMS and it takes me an hour to get back to the print, the layers shift happens as well. I'm afraid it's due to powering off motors after 15 minutes.

DeKra90, 2023-03-07:
I can confirm this too. On one day the manual filament change (with pause) works perfekt and on the next day **after** the update i cant print anything with it. Additionally i can see at which point the layer shift happens. Its right after the extrusion of the new filament and the beginning of the new layer. The head jumps a little bit, perhaps 1mm and then the new layer starts.

EukaDesigns, 2023-03-09:
Based on the g-code, it appears this is for the P1P, and based on @DeKra90's comment, this seems like it might be related to the latest firmware (it worked before, but after the update, it did not work). 

Can @suit confirm which firmware you've used and noticed this issue? For the manual filament change, are you using the unload and/or load button on the P1P?

suit, 2023-03-09:
this happens on 01.02.01.00 - and yes, i'm using the manual load/unload feature introduced in this version

i cannot confirm if it worked as  expected before the firmware update though, sind i've nevertried it

dagnarf, 2023-03-09:
A similar layer shift occurs if the "step recovery on power loss" is turned on and the power is cycled. Perhaps related to the same thing  - the motors powering down,

suit, 2023-03-09:
> A similar layer shift occurs if the "step recovery on power loss" is turned on and the power is cycled. Perhaps related to the same thing - the motors powering down,

in this case the issue also occurs when you change the filament right away - to my informationen, the steppers power down after 15 minutes, i did not exceed this time period, but the shift still occured

Sandermand, 2023-03-14:
Same happens to me on my P1P with no AMS. When it runs out and pauses, the print head moves near the chute and the whole printer starts vibrating violently and making an awful noise. This persists, until i feed new filament into the head and resume the print, at which point the layers shift. Maybe the head "hits" something on its way to the chute or the motors gets jammed ? 

https://user-images.githubusercontent.com/27831056/224965887-1ec9775c-cfc5-4ecc-9ee8-ec40dd31f37e.mp4

Johnnym16, 2023-03-15:
I am having the same issue. The time interval of the filament change does not change the outcome. With the latest firmware update, I get a layer shift every time a filament runs out. 

P1P no AMS

jeppster, 2023-03-24:
I have this issue each time I insert a pause into the print. The first layer after resuming always shifts.

nappyjim, 2023-03-25:
Having this issue as well.  Very annoying.

1) Add pause in Bambu slicer (also tried Orca Slicer)
2) After P1P pauses, use UNLOAD to remove current filament
3) Within 2 minutes, use LOAD to load in new filament
4) Press Resume button on control
5) Watch as my layer shifts......
 
[image]

suit, 2023-03-29:
> Someone "solved" it by adding customer G code found here (https://github.com/EukaDesigns/Bambu_CustomGCode/blob/bfdd8d4e9dce94def22d21b0e4ac8b5e49ebadeb/change_filament_noAMS.gcode_

i'm aware of that - i've just now added a comment on the reddit post of the same guy
https://www.reddit.com/r/BambuLab/comments/11gtlfp/comment/je6zk5y/

problem that i have with this: it is a workaround that is based on an older version of the gcode provided by bambu studio - so there is quite a lot missing

other thing is, this code turns off the fans, heats up the nozzle to the higest temp, cuts and flushes and then pauses the printer - so if you don't change the filament right away, the nozzle stays hot and the fans stay paused

i'm aware that the code is "on your own risk" but, this is not a suggestion that should be made without crosschecking the code

since this is appearantly a printer bug that occurs during homing on the filament unload (not the load) it should be solved on the printer - but the slicer can workaround this

if you don't want to break the AMS in this process, there needs to be condition to check if there is an AMS present or not - and then add a manual pause and a filament cut

those issues are closely related to #1409 and #1408 which can be probably solved with the same solution

a) add a pause on manual filament change
b) unload the filament without homing
c) cool down the nozzle first to new filament temperature before resuming

suit, 2023-03-31:
i found another workaround and can confirm now, that this issue is indeed a result of the UNLOAD function

Steps to avoid the issue

1) add pause at certain layer
2) on pause, manually heat up nozzle to extrusion temperature
3) remove by manually feeding the filament backwards out
4) Feed new filament using the built LOAD function (or manually feed it and hit the extruder button a few times - which is actually faster than the load with lots of random movements that just look fancy and don't do anything)

i printed about 20 testpieces so far, all of them have a perfect layer transition without a shift

if i do the same, and use the UNLOAD function: layershift after layer shift

@SaltWei @sodachen22  is there any news from the printer team about this issue?

nirurin, 2023-04-01:
Just happened to me. 
P1P, brand new (so no worn cutter or anything like that). 

Pause for layer change
Unload
Load
Resume print

Slight shifting of the layer, maybe only half a mm or so but still. 

The best resolution I've seen floated is, instead of using a normal pause, add custom g-code - 
M400 U1
G28 X Y

So it homes after the pause. Am going to test it now to see if it actually works. 

I see there's a long 'custom gcode' solution in this thread but I'm not sure how it's used, and it seems pretty in-depth and convoluted. Will leave it for now.

Spybyte, 2023-05-11:
@SaltWei any news on this from the printer team? It‘s been more than two month since it was reported. Pretty annoying bug.

nirurin, 2023-05-11:
Yeh this was fixed ages ago. Like 3 firmware updates back or something.

Hologos, 2023-05-11:
> Yeh this was fixed ages ago. Like 3 firmware updates back or something.
> […](#)

It was not. They just added feature to load/unload but it happens as well (at least from other bugs).

Spybyte, 2023-05-11:
No, it was not fixed. I have the latest firmware on my P1P and I still have layer shifts. The last one two days ago.

nirurin, 2023-05-11:
Change your pause gcode in your slicer. Seems to work as a fix until bambu
do it properly

Spybyte, 2023-05-11:
I found this thread, read it and thanks to you guys I know how to do this work around. But my question still remains, when will it be fixed? I would say a Bambu printer is a premium consumer device for that price. Because of this I would expect that such bugs would not take more then let’s say two to three weeks to fix. But maybe my expectations are way to high.

nirurin, 2023-05-11:
This is the github for the slicer though, not actually sure it's even on
the main firmware teams radar. Might be worth sending an email to the
actual support team for bambulab and linking this thread there.

HobinBambu, 2023-05-12:
Hi @suit 
    Thank you for your feedback,this is a firmware issue we are fixing and will release soon.A workaround is:
1. Replace the printers/load+unload gcode in studio as below (in the resources/printers folder of the studio installation directory)
[printers.zip](https://github.com/bambulab/BambuStudio/files/11460100/printers.zip)
[image]
2. pause during printing and select ext spool and click "unload"
[image]
3. After unloading is complete, click “load” to load new filament
4. resume print

Tips:

- When the AMS is not connected, the retry button of the studio is not displayed, so if the" load/unload" times out, please click "retry" on the P1P ”Feeding“ tab
[image]

- Because the replacement gcode removes "homing", so if you are not using "load/unload" during printing(such as after power on),please:

1. Use the "load/unload" of APP or printer, they still maintain the original logic;
2. If you use studio, please click the "home" button before "load/unload" as below,
[image]

HobinBambu, 2023-05-12:
@suit @Sandermand 
Thanks.manual load/unload and add "change_filament_noAMS.gcode" are good ways.The above is a way  for still using the "load/unload" feature,and it is applicable to P1P/X1C.

thedodus, 2023-05-29:
I just tried the workaround on my p1p, as in:

1. Filament runs out and printer pauses
2. DID NOT use the "Load" or "Unload" commands
3. Reloaded new filament, heated nozzle to 250 and manually extruded until filament was locked in
4. Hit Resume

And I still got the layer shift of ~2mm to the left.  Did I do something incorrectly above?

Hologos, 2023-06-26:
> Severely disappointed in BambuLabs ... It takes FAR too long to get a suitable fix for the layer shift...
> ...
> The development is far too slow. Bug fixes take forever.

I agree, this is frustrating at least. I myself am waiting for fixes for 2 bugs I've reported almost 3 months ago. 🤦‍♂️

Sandermand, 2023-06-26:
> Severely disappointed in BambuLabs and close to simply sell these printers
> and go back to Prusa.
> It takes FAR too long to get a suitable fix for the layer shift. I've not
> had one such issues on any of my printers in the past.
> Same thing, no matter how a stop is produced (nozzle stuck to a support or
> Spool running out), the resume always ends up with a shift, ruining long
> prints in the process. When I have prints that are detailed (we work with
> art stuff), this is a waste of material, horrible for the environment and
> financially not agreeable.
>
> The development is far too slow. Bug fixes take forever.

The issue was listed as fixed in the latest firmware. You still
experiencing this?

MadKiefer, 2023-06-26:
> The issue was listed as fixed in the latest firmware. You still experiencing this?
> […](#)

Yes. It is still an ongoing issue. Both of our machines have this issue whenever the spool (NOT AMS - yet) is empty.
I even added the 2 parts GCode (the small ones) to prevent the issue.

HobinBambu, 2023-06-26:
@thedodus @MadKiefer sorry for the issue,we fixed the layer shift issue caused when pause the printer and use 'load/unload' commonds in P1P OTA version01.03.00.00 ,and this issue is caused by the 'load/unload' not the 'pasue/resume',If you only use 'pasue/resume' in printing but it still cause layer shift, It is recommended to check whether the PTFE tube or the TYPEC cable will affect the movement of the tool head, or re-tension the belt, if the issue still exists ,we encourage you to reach out to our customer service team [Click here to open a new ticket in our Support Page](https://bambulab.com/en/my/support/tickets),thanks!

closed 2023-06-26 by HobinBambu

namalahideaway, 2024-07-29:
I am having this exact issue even with the most recent firmware

namalahideaway, 2024-07-29:
> @thedodus @MadKiefer sorry for the issue,we fixed the layer shift issue caused when pause the printer and use 'load/unload' commonds in P1P OTA version01.03.00.00 ,and this issue is caused by the [...]

Still having this issue... 

It is not printer mechanics related because if I do a manual pause with the printers physical button it does not cause a layer shift. 

It only happens with the add pause layer in bambu studio. 

What is the fix?