bambulab/BambuStudio #1413: Slicer issues overhang warning for model with no overhang

julie777, 2023-03-06:
**Describe the bug**

I think the image has all the details. The two holes are thru holes with no dimension changes and no hidden overhangs.

[image]

**3mf File for This Bug**

[Winder_hand_crank.zip](https://github.com/bambulab/BambuStudio/files/10893621/Winder_hand_crank.zip)

**To Reproduce**

Steps to reproduce the behavior:
1. Load the model
2. slice it
**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Version 1.4.2.13

tangketan, 2023-03-12:
Around z=6.8mm there are overhangs.
[image]

julie777, 2023-03-19:
Thanks. I didn't see that.

closed 2023-08-02 by tangketan