bambulab/BambuStudio #1414: Bambu Studio 1.5 Parameters not restored

PHIL-IP, 2023-03-06:
**Describe the bug**
Settings like last used filaments and last used profile are not reloaded
Only the last plate is saved/restored.

**To Reproduce**
Just open Bambu Studio, make changes in the filament AMS list and profiles, close Bambu Studio and re-open it. Filaments and profile are the basics settings.

**Expected behavior**
When we open Bambu Studio reload the last filaments used and reload the last profile used (not the basic 0.20mm standard @BBL X1C profile)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10 Pro
 - Version of Bambu Studio : 1.5.0

Sorry for my english...

lanewei120, 2023-03-06:
could you give a gif or a short video to show the detailed steps?
we should load the profiles used last time when re-launch slicer

Ali229, 2023-03-06:
Also having the issue GIF and log attached:
[image]

[debug_Mon_Mar_06_08_48_42_18240.log.zip](https://github.com/bambulab/BambuStudio/files/10898854/debug_Mon_Mar_06_08_48_42_18240.log.zip)

PHIL-IP, 2023-03-06:
GIF and log file : 
[image]
[debug_Mon_Mar_06_20_22_40_5872.log](https://github.com/bambulab/BambuStudio/files/10902132/debug_Mon_Mar_06_20_22_40_5872.log)

bambu123, 2023-03-07:
Please also help private the BambuStudio.conf file. Be sure to select some preset and close BambuStudio before sending the file.

PHIL-IP, 2023-03-07:
Hi, my settings just before closing Bambu Studio :
[image]
... and my BambuStudiuo.conf file :
[BambuStudio.zip](https://github.com/bambulab/BambuStudio/files/10906088/BambuStudio.zip)

bambu123, 2023-03-07:
Will be fixed in the 1.5 patch version.

PHIL-IP, 2023-03-07:
Good news.
Thank you

closed 2023-03-07 by PHIL-IP