bambulab/BambuStudio #1416: Focus zoom on selected object

noossyy, 2023-03-07:
**Is your feature request related to a problem? Please describe.**

Without a focus zoom on an object, your can’t properly rotate around and object after having zoomed on it. 

**Describe the solution you'd like**
Replicate the same behavior as the Z hotkey in superslicer or F hotkey in Autodesk Maya when object is selected. The camera will zoom close to it and center focus on it so you can properly rotate around it. 

**Describe alternatives you've considered**
Having a hotkey to do that like in most 3D animation software 

**Additional context**
Add any other context or screenshots about the feature request here.

BowieInc, 2023-09-27:
Yes! How can I add my support for this shortcut addition? It is much needed.

kertgartner, 2023-11-24:
YES PLEASE! Would love to see this added!

Thelvaen, 2024-01-04:
I'd love to see this feature as well.

CaidenKehrer, 2024-08-30:
I would also really appreciate this feature. It is nearly impossible to get angles right to view some objects .

travrob1, 2025-01-04:
Yes please!!! That woulds same me so much time.  Thanks

ctkjedi, 2025-06-26:
Another vote for this! Zoom to mouse is fine, but the moment you rotate your view...

StarDreamIX, 2026-11-03:
Please add this!! - Just like in Blender when you press Del key or full stop etc it should zoom in on selected object! Please add this!