bambulab/BambuStudio #1417: Added printing speed in Bambu Studio.

PHIL-IP, 2023-03-07:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Addition of a menu allowing to launch a printing directly in silent mode or sport etc...
At the moment you have to start printing first and then change the value of the printing speed.

**Describe alternatives you've considered**
None except to start the printing and then go to change the speed of it

Somerandomgithugguy, 2023-04-17:
Yes we really need this, it is so annoying to have to manually change it every time after starting the print!

Somerandomgithugguy, 2023-04-17:
Same issue as raised here #1290  and #1419

daliudzius, 2023-05-03:
This feature request would be to slice a project with easy speed settings included - for example slicing with "silent" speed already set in Bambu Studio so we don't have to wait for the print to start to toggle speed settings. 

Both issues [#1290](https://github.com/bambulab/BambuStudio/issues/1290) and [#1419](https://github.com/bambulab/BambuStudio/issues/1419) relate only to adding additional speed options.

aliasmeum, 2023-10-24:
Before version 1.7 it was possible to change to silent mode as soon as the slicer jumped to the device page. Now you have to wait (I believe, until the bed is heated?). If you try to change the speed before this, it just jumps back to 100%. Having to sit around waiting for the bed to heat to be able to change the speed is ridiculously annoying. Could ability to choose speed in the "Send to printer" window be an option? Preferably by setting the percentage.

Somerandomgithugguy, 2024-01-02:
It is so sad that this is not implemented yet. :(

anoblet, 2024-02-09:
Is there any update on this?

Somerandomgithugguy, 2024-02-09:
> Is there any update on this?

Sadly not...

sodachen22, 2024-02-11:
We got this feature request and will evaluate it. Thanks you!

kzlsnn, 2024-08-11:
It would be great if this feature was added thx.