bambulab/BambuStudio #1423: Device/Print Options has poor wording

julie777, 2023-03-07:
**Describe the bug**
I believe that the checkbox label should read "auto-recover from step loss" and not "recovery".

[image]

Also, it would be nice if there was a tooltip or some kind of explanation about it.
Because I don't completely understand it, I don't know if the button should be enabled when printing or when not printing. Currently it is always enabled.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

StoneLiBambulab, 2023-03-30:
Hi @julie777 ,
    i will be fixed in next version.

Thanks!

Stone Li

StoneLiBambulab, 2023-04-24:
[image]

included in 1.6.0, closed.

closed 2023-04-24 by StoneLiBambulab