bambulab/BambuStudio #1424: P1P device start printing dialog is missing choice to skip resonance testing.

julie777, 2023-03-07:
**Describe the bug**

There is a calibration dialog that allows choosing both bed leveling and resonance testing. The start print dialog only allows skipping bed leveling.

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. select "Print Plate"
2. the start printing dialog will appear
3. it is missing the skip resonance testing choice

**Expected behavior**

A clear and concise description of what you expected to happen.

**Screenshots**

[image]

[image]

**Desktop (please complete the following information):**
 - OS:  Windows 10
 - Version 1.4.2.13

lanewei120, 2023-03-14:
hello
the "resonance testing" is not a feature need to run frequently
we won't run "resonance testing" when send a print

so there is no need to add the option to skip it

julie777, 2023-03-19:
Isn't the resonance test what is happening during start up when the print head just starts vibrating?

lanewei120, 2023-03-20:
no, that is not the resonance test
it won't be done at each print

julie777, 2023-03-20:
@lanewei120 
Okay. Thank you for your help.

Would you mind telling me what the purpose of the following is during the startup?
1. The bed vibrates. I watched an entire startup with bed calibration and at many of the calibration points the bed vibrated up and down.
2. There was also a point where the print head was over the center of the bed and just did a short but extreme vibration. What is that for?

I will close the issue after your response.

lanewei120, 2023-03-21:
1. it is a necessary process for bed calibration, however I don't know the details
2. it is doing a fast mechanical resonance mode checking , only takes a few seconds , so we think it is acceptable for each startup

closed 2023-03-22 by julie777