bambulab/BambuStudio #1425: Name of item being printed is truncated

julie777, 2023-03-07:
**Describe the bug**

The actual name show in the Bambu Studio Title bar is "Print plate Winder and Carriage" but the printing status cuts of part of the title. I don't know if it is based on length or number of characters.

[image]

**3mf File for This Bug**
If it is related to slicing, please append the 3mf file. It could be extremely helpful to solve the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS, Windows]
 - Version [e.g. 22]

walterwongbbl, 2023-04-07:
Hi.
Please confirm if the name on the printer is correct

walterwongbbl, 2023-10-13:
Please download and use the latest version of Studio

closed 2023-10-13 by walterwongbbl