bambulab/BambuStudio #1426: Linux: Live Stream Video Playback fails without ALSA present

norpol, 2023-03-07:
**Describe the bug**

Without ALSA, hitting play at the the Live Stream yields to a popup `Media playback error: Could not open audio device for playback.` and no video starting.

**To Reproduce**

Steps to reproduce the behavior:
1. Use Pulseaudio or pipewire without ALSA compatibility enabled (or run desktop without audio stack)
2. Click on Video Playback
3. See error

**Expected behavior**

Gstreamer should only playback a video stream and never try to access audio. At the moment, when you're clicking playback of the live stream, gstreamer tries to access several alsa configuration files and ultimately fails if no alsa device is present.

Playing the video shows that gstreamer doesn't need to initialize the audio playback in first place, as no audio is playing.

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Linux, Wayland, SwayWM, PipeWire
 - Version: 1.4, 1.5, ..

bambu123, 2023-03-10:
Thank you for your report!

bambu123, 2023-06-06:
@norpol
I have a patch, can you help test it?
Do you build the app image by yourself?
[a65ecc8.diff.zip](https://github.com/bambulab/BambuStudio/files/11663690/a65ecc8.diff.zip)

norpol, 2023-06-06:
Hey @bambu123 Yes I'm building myself with NixOS. I'll test it asap. Thank you a ton for the change already! :)

bambu123, 2023-06-07:
> Hey @bambu123 Yes I'm building myself with NixOS. I'll test it asap. Thank you a ton for the change already! :)

Thank you for your cooperation!

norpol, 2023-06-07:
@bambu123 I'll give it a shot, takes a while to build but I think it might be blocked to test it right now because of https://github.com/bambulab/BambuStudio/issues/1843 but I'll give it a try.

bambu123, 2023-06-08:
> Well done @bambu123, your patch applied to `v01.05.00.61` fixes the problem. I've verified by disabling pipewire-alsa and starting the previous build today video playback would still fail, applying the patch the new one displays the video correctly with no error. Thank you.

Great! Thank you a lot.

closed 2023-07-11 by bambu123