bambulab/BambuStudio #1430: First layer travel speed

npoulos, 2023-03-08:
**Is your feature request related to a problem? Please describe.**
Can you please allow for the ability to add a separate travel speed for the first layer. The first layer needs to be printed slow for good adhesion but the travels kind of mess it up.

SaltWei, 2023-03-10:
[image]
Do you mean the travel speed of first layer should be slower？

npoulos, 2023-03-10:
Yea but with an input to define what the travel speed is for the first layer

SaltWei, 2023-04-13:
We have tested lower first layer travel speed has no improvement for bed adhesion, but lower first layer acceleration helps.
If you meet bed adhesion issue. Please share what filament you are using.

QingZhangBambu, 2023-06-13:
we will add that soon

QingZhangBambu, 2025-05-30:
It has been resolved. Closing this issue.

closed 2025-05-30 by QingZhangBambu