bambulab/BambuStudio #1431: Beta 1.5.0 can't install network plugin on a domain PC on Windows 11

eZtaR1, 2023-03-08:
**Describe the bug**
I can't install the new 1.5.0 Beta on my work PC as the network plugin fails in two ways.

**To Reproduce**
Steps to reproduce the behavior:
1. Use a domain PC on Win 11
2. Install  Bambu_Studio_win_public-v01.05.00.59-20230301140620.exe via your elevated credentials
3. Open bambu studio in your non-elevated user.
4. Get the install error (it doesn't ask for elevated credentials)
[image]

1. Use a domain PC on Win 11
2. Install  Bambu_Studio_win_public-v01.05.00.59-20230301140620.exe via your elevated credentials
3. Open bambu studio via "Run as admin"
4. Popup saying "Microsoft Edge can't read or write to it's data-folder C:\users\<username>.domain\AppData\Local\BambuStudio\EBWebView"
5. Get blank setup frame forcing you to end task.

[image]
[image]

**Expected behavior**
Installation of network plugin

**Desktop (please complete the following information):**
 - Windows 11 Enterprise Evaluation
 - Build 2261.ni_release.220506-1250

miljbee, 2023-03-09:
Hello,

same problem here with 01.05.00.60 on windows 10.

miljbee, 2023-03-09:
found this in the log:
```
2023-03-09 22:44:30.354653[Thread 0x00006074]:[download_plugin] get_url = https://public-cdn.bambulab.com/upgrade/studio/plugins/01.05.00.10/win_01.05.00.10.zip
2023-03-09 22:44:30.669515[Thread 0x00006074]:UpgradeNetworkJob: percent = 10msg = Downloading
2023-03-09 22:44:30.669515[Thread 0x00006074]:[download_plugin 2] progress: 10
2023-03-09 22:44:30.774136[Thread 0x00006074]:UpgradeNetworkJob: percent = 20msg = Downloading
2023-03-09 22:44:30.774136[Thread 0x00006074]:[download_plugin 2] progress: 20
2023-03-09 22:44:30.898004[Thread 0x00006074]:UpgradeNetworkJob: percent = 30msg = Downloading
2023-03-09 22:44:30.898004[Thread 0x00006074]:[download_plugin 2] progress: 30
2023-03-09 22:44:31.005060[Thread 0x00006074]:UpgradeNetworkJob: percent = 40msg = Downloading
2023-03-09 22:44:31.005060[Thread 0x00006074]:[download_plugin 2] progress: 40
2023-03-09 22:44:31.286969[Thread 0x00006074]:UpgradeNetworkJob: percent = 50msg = Downloading
2023-03-09 22:44:31.286969[Thread 0x00006074]:[download_plugin 2] progress: 50
2023-03-09 22:44:31.286969[Thread 0x00006074]:[download_plugin 2] completed
2023-03-09 22:44:31.343818[Thread 0x00006074]:UpgradeNetworkJob: percent = 80msg = Downloading
2023-03-09 22:44:31.344819[Thread 0x00006074]:[install_plugin] enter
2023-03-09 22:44:31.344819[Thread 0x00006074]:UpgradeNetworkJob: percent = 50msg = Installing
2023-03-09 22:44:31.344819[Thread 0x00006074]:[install_plugin]: 1499, got 2 files
2023-03-09 22:44:31.345816[Thread 0x00006074]:[install_plugin]Archive read exception:boost::filesystem::remove: Access denied [system:5]: "C:\Users\zzzzzzzzz\AppData\Roaming\BambuStudio\plugins/BambuSource.dll"
```
downloaded the file from https://public-cdn.bambulab.com/upgrade/studio/plugins/01.05.00.10/win_01.05.00.10.zip
put it manually in the plugins folder. Solved.
I can't tell why there has been an access denied in the log, I replaced the files from my explorer without elevation ...

eZtaR1, 2023-03-10:
Installed via  Bambu_Studio_win_public-v01.05.00.60-20230309155250.exe on a freshly installed machine today to great success!
[image]

bjornsyse, 2023-03-11:
I'm gettting this error too. Manually putting the files from https://public-cdn.bambulab.com/upgrade/studio/plugins/01.05.00.10/win_01.05.00.10.zip into the plugins folder did not solve the issue for me.

lanewei120, 2023-03-14:
it seems previous plugins can not be removed by studio this time.

we can check the files in the plugins folder
[image]

and delete it manually before install

lanewei120, 2023-03-14:
> **Describe the bug** I can't install the new 1.5.0 Beta on my work PC as the network plugin fails in two ways.
> 
> **To Reproduce** Steps to reproduce the behavior:
> 
> 1. Use a domain PC on Win 11
> 2. Install  Bambu_Studio_win_public-v01.05.00.59-20230301140620.exe via your elevated credentials
> 3. Open bambu studio in your non-elevated user.
> 4. Get the install error (it doesn't ask for elevated credentials)
>    [image]
> 5. Use a domain PC on Win 11
> 6. Install  Bambu_Studio_win_public-v01.05.00.59-20230301140620.exe via your elevated credentials
> 7. Open bambu studio via "Run as admin"
> 8. Popup saying "Microsoft Edge can't read or write to it's data-folder C:\users<username>.domain\AppData\Local\BambuStudio\EBWebView"
> 9. Get blank setup frame forcing you to end task.
> 
> [image] [image]
> 
> **Expected behavior** Installation of network plugin
> 
> **Desktop (please complete the following information):**
> 
> * Windows 11 Enterprise Evaluation
> * Build 2261.ni_release.220506-1250

do you still have the blank frame issue caused by "Microsoft Edge can't read or write to it's data-folder C:\users<username>.domain\AppData\Local\BambuStudio\EBWebView"

bambu123, 2023-03-15:
@bjornsyse What's the detail when it fails? Please send us the logs from 'Help > Configuration Folder'. Also, check whether you can delete the Dlls in the plugins folder when it fails.

bambu123, 2023-03-15:
We are not going to focus on the Edge problem because it is not the normal way of use to start as an Admin.

miljbee, 2023-03-21:
> @miljbee <https://github.com/miljbee> @bjornsyse
> <https://github.com/bjornsyse> Are your problem still exist?

Solved by deleting the old plugin manually.

bjornsyse, 2023-03-21:
I'm good too

eZtaR1, 2023-04-05:
> > **Describe the bug** I can't install the new 1.5.0 Beta on my work PC as the network plugin fails in two ways.
> > **To Reproduce** Steps to reproduce the behavior:
> > 
> > 1. Use a domain PC on Win 11
> > 2. Install  Bambu_Studio_win_public-v01.05.00.59-20230301140620.exe via your elevated credentials
> > 3. Open bambu studio in your non-elevated user.
> > 4. Get the install error (it doesn't ask for elevated credentials)
> >    [image]
> > 5. Use a domain PC on Win 11
> > 6. Install  Bambu_Studio_win_public-v01.05.00.59-20230301140620.exe via your elevated credentials
> > 7. Open bambu studio via "Run as admin"
> > 8. Popup saying "Microsoft Edge can't read or write to it's data-folder C:\users.domain\AppData\Local\BambuStudio\EBWebView"
> > 9. Get blank setup frame forcing you to end task.
> > 
> > [image] [image]
> > **Expected behavior** Installation of network plugin
> > **Desktop (please complete the following information):**
> > 
> > * Windows 11 Enterprise Evaluation
> > * Build 2261.ni_release.220506-1250
> 
> do you still have the blank frame issue caused by "Microsoft Edge can't read or write to it's data-folder C:\users.domain\AppData\Local\BambuStudio\EBWebView"

Sorry, I totally overlooked you message - but as i replied further up, my problems evaporated with the latest release exe.

bambu123, 2023-04-06:
@eZtaR1 Logs such as @miljbee‘s ones may help. You can find log from the Menu 'Help > Open Configuration Folder"

Chry007, 2023-05-17:
For me, MS Teams blocked a bambu dll File. Closing MS Teams and then installing worked:
2023-05-17 12:03:05.712332[Thread 0x00007278]:[install_plugin]Archive read exception:boost::filesystem::remove: Access Denied [system:5]: "C:\Users\....\AppData\Roaming\BambuStudio\plugins/BambuSource.dll"

bambu123, 2023-05-30:
@eZtaR1 Yes, same as #1748

bambu123, 2024-02-06:
It seems that all problems have been solved, close the issue.

closed 2024-02-06 by bambu123

Hackers59551, 2026-08-19:
У меня после обновления на версию 2.8.2.60, не скачивает плагин, хотя я не менял никаких настроек в сети и впн не ставил, так ещё и по инструкции в ручную установку не находит такой файл как будто его и нет вообще